hybridgroup/gobot #315: Platform request: LED strip (e.g LDP8806)

Inux, 2016-09-26:
I would like to implement an LED strip driver in the GPIO package. But I have some questions which have to be answered before I can start.

The LDP8806 is basically an SPI slave but I would like to implement it with normal GPIO's instead of the HW SPI. Therefore I thought it would make sense to add a new interface into gpio.go. 
I would like to implement it so that other led strips can be integrated easily.

To my questions:
Does it makes sense to implement a new interface in gpio.go for SW SPI? (SW SPI means SPI with normal GPIO's) or is there an other way?
Are there already plans for led strip implementation?

deadprogram, 2016-10-03:
Hi, @Inux I think we need a standard SPI interface. If there is a software SPI implementation, also, that would be great. 

Then whatever SPI drivers can use that interface, such as the LED strip you mention.

Does this make sense?

gen2thomas, 2022-12-08:
As far as I can see, the "warthogs gpiod" implementation brings such [SPI implementation](https://github.com/warthog618/gpiod/blob/master/spi/spi.go) using the GPIOs. So I will prepare the possibility to switch between different implementations and maybe already integrate the new implementation, since we use the library already for character device Kernel ABI.

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas