hybridgroup/gobot #317: Edison documentation

devfubar, 2016-10-06:
I am new to gobot and I was trying to get my Intel Edison working. I do not have the arduino breakout board and spent some time figuring our that you make an assumption that it is present when using the Edison. I am interested in issue #262 however I think that the documentation should be updated to state the Arduino breakout board assumption to prevent people scratching their heads for as long as I did.

deadprogram, 2016-10-07:
Hi @devfubar that is no doubt true. Once the current refactoring in PR has been merged into master, I expect to add some code to handle the Arduino as well as the mini-breakout board, as well as to show other contributors how to more easily add other Edison breakout boards.

devfubar, 2016-10-07:
Thanks @deadprogram for getting back to me.  I was just interest to see if this was still the plan as the last comment on #262 was in Feb. I raised this issue as an interim solution to prevent people from wondering why the basic example wasn't working.

I'm happy to contribute so some sort of guide that you mentioned would be very useful.

deadprogram, 2016-11-01:
The `dev` branch now has all of these changes, thank you very much @devfubar for your awesome contributions on this.

closed 2016-11-01 by deadprogram