hybridgroup/gobot #319: go get failing due to missing platform libs in dev branch

conoro, 2016-10-10:
I'm trying to install Gobot on a C.H.I.P and I'm failing on the first hurdle. 

```
go get -d -u github.com/hybridgroup/gobot/...
```

gives

```
cannot find package "github.com/hybridgroup/gobot/platforms/i2c"
```

i2c and gpio seem to be missing on the default dev branch in GitHub but they are on master. Is there some trick to getting master other than checking everything out manually? Thanks.

deadprogram, 2016-10-10:
Hi @conoro 

both the `i2c` and `gpio` packages are now under `/drivers` in the `dev` branch, but this seems to be an unforeseen consequence of Github, having set the default branch to `dev`, and how Golang gets packages.

What I thought was happening was Golang would simply always get the `master` branch, sinec that used to be the default behavior. My knowledge on that subject was way out of date, it would seem. What appears to actually be happening is that Github serves up the default branch https://github.com/golang/go/issues/9032

Ouch, my bad. So much for our careful release planning.

I think the best thing to do for now, is switch the default branch back to master. Hopefully that will give a chance to complete the refactoring work taking place right now, for which the docs are not yet complete.

conoro, 2016-10-10:
@deadprogram Thanks! I also found a quick workaround which was just to use govendor like so:

```
govendor init
govendor fetch github.com/hybridgroup/gobot/...@master
```

I was then able to build and run the sample app

deadprogram, 2016-10-10:
Did you obtain the code for that sample from the web site? I ask because the code in the repo itself has been modified for compatibility with the latest changes.

conoro, 2016-10-10:
Yeah I just used the code from the bottom of this page: https://gobot.io/documentation/platforms/chip/

deadprogram, 2016-10-11:
I cut a release yesterday (0.13.0) with the code up the point I merged the new refactored branch on Monday, and then switched back the master branch to be the default branch. This way, anyone who was tracking the default up to that point has a master branch to match.

I will close this issue now, thanks for pointing it out @conoro

closed 2016-10-11 by deadprogram