hybridgroup/gobot #324: Initial level question

gpaixao, 2016-10-21:
Hi,

I'm starting now with arduino and already developing in Golang, my problem is maybe easy.

My `func`work don't work 😄 

```go
package main

import (
	"fmt"

	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/firmata"
	"github.com/hybridgroup/gobot/platforms/gpio"
)

func main() {
	gbot := gobot.NewGobot()

	firmataAdaptor := firmata.NewFirmataAdaptor("arduino", "/dev/cu.usbmodem1421")
	sensor := gpio.NewAnalogSensorDriver(firmataAdaptor, "sensor", "0")

	work := func() {
		fmt.Println(sensor.Read())
	}

	robot := gobot.NewRobot("bot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{sensor},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}

```

Anybody, can you help me ?

tks

deadprogram, 2016-10-21:
Hi, @gpaixao 

I ran your code against the current release, and received the following output:

```
$ go run examples/firmata_analog.go 
2016/10/21 08:40:44 Initializing Robot bot ...
2016/10/21 08:40:44 Initializing connections...
2016/10/21 08:40:44 Initializing connection arduino ...
2016/10/21 08:40:44 Initializing devices...
2016/10/21 08:40:44 Initializing device sensor ...
2016/10/21 08:40:44 Starting Robot bot ...
2016/10/21 08:40:44 Starting connections...
2016/10/21 08:40:44 Starting connection arduino on port /dev/ttyACM0...
2016/10/21 08:40:49 Starting devices...
2016/10/21 08:40:49 Starting device sensor on pin 0...
2016/10/21 08:40:49 Starting work...
0 <nil>
```

It ran, and printed a single value from the sensor, in this case 0. Is that not what you expected? If you want to print the value over and over, you need to do that in your work. For example, to read and print the value every 1 second:

```go
package main

import (
	"fmt"
	"time"

	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/platforms/firmata"
	"github.com/hybridgroup/gobot/platforms/gpio"
)

func main() {
	gbot := gobot.NewGobot()

	firmataAdaptor := firmata.NewFirmataAdaptor("arduino", "/dev/cu.usbmodem1421")
	sensor := gpio.NewAnalogSensorDriver(firmataAdaptor, "sensor", "0")

	work := func() {
		gobot.Every(1*time.Second, func() {
			fmt.Println(sensor.Read())
		})
	}

	robot := gobot.NewRobot("bot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{sensor},
		work,
	)

	gbot.AddRobot(robot)

	gbot.Start()
}
```

Hope that helps!

gpaixao, 2016-10-23:
Don't work, my log:

closed 2016-10-24 by gpaixao