hybridgroup/gobot #325: Request: PIR Motion Sensor

rraulinio, 2016-10-22:
I recently got a motion sensor for my Arduino Uno and I'm interested in programming it via Gobot, because I've done everything so far in Gobot.
The sensor is [https://www.google.ro/search?q=arduino+motion+sensor&biw=1280&bih=593&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjYzdPtuezPAhXHJSwKHa5lDkcQ_AUIBigB](url).

Looking forward for an answer!

deadprogram, 2016-10-22:
Hi @rraulinio please take a look at this feature branch with initial PIR sensor support: https://github.com/hybridgroup/gobot/tree/feature/devices/pirmotion

You would use it as follows:

```go
package main

import (
	"github.com/hybridgroup/gobot"
	"github.com/hybridgroup/gobot/drivers/gpio"
	"github.com/hybridgroup/gobot/platforms/firmata"
)

func main() {
	firmataAdaptor := firmata.NewAdaptor("/dev/ttyACM0")

	sensor := gpio.NewPIRMotionDriver(firmataAdaptor, "5")
	led := gpio.NewLedDriver(firmataAdaptor, "13")

	work := func() {
		sensor.On(gpio.MotionDetected, func(data interface{}) {
			led.On()
		})
		sensor.On(gpio.MotionStopped, func(data interface{}) {
			led.Off()
		})
	}

	robot := gobot.NewRobot("motionBot",
		[]gobot.Connection{firmataAdaptor},
		[]gobot.Device{sensor, led},
		work,
	)

	robot.Start()
}
```

Note that this branch is based on the `dev` branch, and so will not be compatible with the current `master`. Please give it a try if you have a chance.

rraulinio, 2016-10-22:
You forgot to put the gpio folder in platforms/ ?
Also, here is the output I receive when I try to compile a simple blink program.
[https://s17.postimg.org/ineo61zv3/Untitled.png](https://s17.postimg.org/ineo61zv3/Untitled.png)

rraulinio, 2016-10-22:
@deadprogram I will test the PIR sensor on monday and come back with a later edit. Tested the branch files, everything works fine (after editing main.go as you said). Also, when will it be available on master? I'm asking because I need to know if it's worth to commit to git or not. :)

rraulinio, 2016-10-26:
Nope, the function doesn't work. The sensor is not detecting anything or it isn't starting, don't know exactly. Please help me asap!

deadprogram, 2016-10-30:
Hi @rraulinio I had to make another commit to correct an error in the original code, and I just tested it with my HC-SR501 sensor. Please give another try when you get a chance.

rraulinio, 2016-10-30:
Are you sure that you commited the right code? It still doesn't work and you forgot that fmt.println in an infinite loop...

deadprogram, 2016-10-30:
The correct code is now checked in. This is my output:

```
$ go run examples/firmata_pir_motion.go 
2016/10/30 22:05:21 Initializing connections...
2016/10/30 22:05:21 Initializing connection Firmata ...
2016/10/30 22:05:21 Initializing devices...
2016/10/30 22:05:21 Initializing device PIRMotion ...
2016/10/30 22:05:21 Initializing device LED ...
2016/10/30 22:05:21 Robot motionBot initialized.
2016/10/30 22:05:21 Starting Robot motionBot ...
2016/10/30 22:05:21 Starting connections...
2016/10/30 22:05:21 Starting connection Firmata on port /dev/ttyACM0...
2016/10/30 22:05:26 Starting devices...
2016/10/30 22:05:26 Starting device PIRMotion on pin 5...
2016/10/30 22:05:26 Starting device LED on pin 13...
2016/10/30 22:05:26 Starting work...
motion-detected
motion-stopped
motion-detected
motion-stopped
motion-detected
```

rraulinio, 2016-10-30:
Mnope, still not working. Not detecting any motion at all, don't even know if it's calibrating or something... Be sure that everything is plugged corectly, because as I said, it works runned by Arduino compiler. I'm running this very basic example and nothing... led on 13, sensor on 7... nothing. I don't know...

deadprogram, 2016-10-30:
I know my HW is plugged in correctly, and as you can see form the above output is working. There is no calibration on the sensor from software, it just does it's thing.

rraulinio, 2016-10-31:
Mnope, tested again via Arduino's compiler and works perfectly.
Reuploaded Firmata.
Testing via Gobot, not working. I'm out of ideas...

rraulinio, 2016-10-31:
https://s17.postimg.org/rdo910isf/image.png

deadprogram, 2016-10-31:
I'd check your version of Firmata. Where did you obtain it from? Also verify your pin, my test used digital pin 5, you mentioned you were using digial 7. Also make sure you are using correct 5v voltage.

I used an Arduino Uno. Are you using some other Arduino variant?

In any case, check all the above, since we know the code itself "works".

rraulinio, 2016-10-31:
Firmata is at ir's latest version, from October 2016, being updated in Arduino's compiler and then uploaded by me.
I'm sure that the cables are plugged in ok and the sensor works, because as i said, if i run "pir motion example on arduino compiler" it works perfectly.
I'm using Uno, compatible with Arduino... it's the same thing.

rraulinio, 2016-10-31:
I really can't see why it does not work, mostly because it worked for you. Are you sure that you uploaded all the corrections for the code? Maybe you forgot something, don't know.

deadprogram, 2016-10-31:
Did you verify that the Gobot `firmata_blink.go` program works as you expect? Try testing that on the Gobot `dev` branch as sanity check.

I use ConfigurableFirmata with most everything activated (DigitalIO, AIO, I2C), usually uploaded via the latest version of Gort (https://github.com/hybridgroup/gort). Also have tested using Arduino101 using the Firmata uploaded via IMRAA.

If not, something else must be different on your setup. Are you just running the https://github.com/hybridgroup/gobot/blob/feature/devices/pirmotion/examples/firmata_pir_motion.go program, or something else?

rraulinio, 2016-10-31:
Everything runs properly. 
In parallel, I run 3 leds and a buzzer from the breadboard, so yes, everything is allright at this point. 
I'm running that example, yes.

deadprogram, 2016-10-31:
This is my hw setup:
[image]

rraulinio, 2016-10-31:
Uploaded firmata via Gort, retested...  still nothing.
Tested in that very basic example... nothing.
It can't be possible that the configuration could have something, because in this case, running it via Arduino's compiler wouldn't work at all, in the first place. It's crystal clear that the problem is either at my version of firmata (don't think so, it's the last one), either on something related to gobot. Please post 2-3 more pictures, with the connections of wires and all, so I can make a check with yours.

deadprogram, 2016-10-31:
When you said "Tested in that very basic example... nothing" did you mean that the `firmata_blink.go` example did not work for you?

deadprogram, 2016-10-31:
Here are my connections to the HC-SR501:

[image]

The OUT pin on the sensor is connected to the digital pin 5.

rraulinio, 2016-10-31:
Yes, very basic example = firmata_blink.go.
I'm gonna get back at it in a few hours (it's morning here in Romania and I need to go away from home).

deadprogram, 2016-10-31:
Did the firmata_blink.go work, or not work?

rraulinio, 2016-10-31:
Not work. Initialising and starting sensor, but that's it, no detection further on.

deadprogram, 2016-10-31:
I was asking about the firmata_blink.go, not the firmata pir.

rraulinio, 2016-10-31:
Sorry, I was in s hurry, didn't pay full attention. Yes, it worked perfectly.

rraulinio, 2016-10-31:
I will get back at in about 6-7 hours. I think i'm gonna record a short video with the configuration and with the startup and email it to you on your account's address. Is that ok?

deadprogram, 2016-10-31:
Sounds great, thanks.

rraulinio, 2016-10-31:
No, I thank you. Than, talk to you via email and i will get back here with a new post with the solution when it's done (for others in the same situation).

rraulinio, 2016-10-31:
Email sent, check your inbox.

deadprogram, 2016-11-01:
You probably want to set the jumper to "auto-reset" right now looks like you are set to "no reset": https://www.mysensors.org/build/motion

The Arduino code from what I could tell in the video looks opposite to what I would expect. The HC-SR501 goes HIGH when motion is detected, and LOW when no motion.

rraulinio, 2016-11-01:
So what should I do? I don't understand.

deadprogram, 2016-11-01:
Switch your jumper to the "auto-reset" position, I suppose.

rraulinio, 2016-11-01:
Switched, no change... still not working. Sometimes (not always), if I replug the red wire (GRD), it goes on and detects, and than goes off. After that... nothing. I tried every single posibility, with the jumper on L and on H, nothing.

deadprogram, 2016-11-01:
I have a bunch of these same sensors, they are very slow to reset (needs at minimum 5 seconds) but generally reliable if you set the calibration. You for sure need auto reset or it will only trigger once.

I have tested successfully with a real Arduino Uno, a clone Arduino Uno, and an Arduino 101. I've also tested with a couple of different pins.

In any case, check your sensor, or your cables, or try another Arduino. Seems like everything is working as expected to me.

rraulinio, 2016-11-01:
Ok, I will see, than. I will try to figure it out. Thanks for all the help,
I really thank you. Hope this will help others, too.

deadprogram, 2016-11-01:
I have merged this branch into the dev branch, so please continue to test from there.

2 other questions:

- which brand of Arduino are you using?
- what version of Golang are you running.

Please let me know anything new you might discover.

rraulinio, 2016-11-01:
Uno R3, made like Arduino. Is a replica of Arduino, it's around $6-7 here and does the same thing as an original one. :)) I've tested a lot of algorithms on it, made by me or not, and it works very well. I'm running Go 1.7.3 on Win.

rraulinio, 2016-11-01:
No, I just can't make it work. I've tried everything, really everything. I doesn't work with Gobot... only with that Arduino's Compiler. I need to give up for now, I can't focus on this anymore. I will get back at it in a few weeks or something like this. Thanks again for all your help!

deadprogram, 2016-11-04:
This device is now in the 'dev' branch. I am closing this issue, please reopen if needed, and thanks to @rraulinio for your work on this.

closed 2016-11-04 by deadprogram