hybridgroup/gobot #335: go installation problem, osx 10.12.1

alexander-yun, 2016-11-07:
go get -d -u github.com/hybridgroup/gobot/...                                                 1:11
package github.com/hybridgroup/gobot/...: mkdir /home/username: operation not supported

deadprogram, 2016-11-07:
Hi, @alexander-yun 

What does the "1:11" in your installation signify? In any case, please make sure you have followed the official Golang install instructions for OSX https://golang.org/doc/install

Seems like this is a problem with your Golang installation, not with Gobot itself.

Hope that helps!

alexander-yun, 2016-11-08:
Hi, deadprogram
1:11 is time when I run the command "go get -d -u github.com/hybridgroup/gobot/..."
Your answer helped me. 
In  Golang install instructions official page I review again environment variables and find mistake.
Now install works.

Thank you.

closed 2016-11-08 by alexander-yun