hybridgroup/gobot #339: MQTT Paho location changed

juliancheal, 2016-11-16:
Hey @deadprogram,

It seems that `git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git` is no longer available via `go get`

I tried just changing the dependency to `github.com/eclipse/paho.mqtt.golang`, but I think that's a newer version and I just got the following errors.

```
../.go/src/github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go:39: cannot use mqtt.NewClient(createClientOptions(a.clientID, a.Host, a.username, a.password)) (type mqtt.Client) as type *mqtt.Client in assignment:
	*mqtt.Client is pointer to interface, not interface
../.go/src/github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go:40: a.client.Connect undefined (type *mqtt.Client is pointer to interface, not interface)
../.go/src/github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go:50: a.client.Disconnect undefined (type *mqtt.Client is pointer to interface, not interface)
../.go/src/github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go:66: a.client.Publish undefined (type *mqtt.Client is pointer to interface, not interface)
../.go/src/github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go:75: a.client.Subscribe undefined (type *mqtt.Client is pointer to interface, not interface)
```

They maybe unrelated, but I don't have time right now to have a deeper look. Happy to once I get some time.

deadprogram, 2016-11-17:
Use the dev branch for now, that was addressed a bit ago.

juliancheal, 2016-11-17:
@deadprogram Thanks Ron :)

closed 2016-11-17 by juliancheal

georgik, 2016-11-18:
Would it be possible to add this simple patch to master?
It took me about hour to find out where is the problem and that there is no such repository at eclipse.org.

Manual workaround is to update: github.com/hybridgroup/gobot/platforms/mqtt/mqtt_adaptor.go
And change line 4 to: "github.com/eclipse/paho.mqtt.golang"

Thank you for considering such a change.

deadprogram, 2016-11-24:
Hi @georgik that fixes the import, however not the code itself. Seems like we need to get that release 1.0 out!

georgik, 2016-11-25:
Thank you for explanation. I also noticed that Gobot web site contains examples from master version and they are not always compatible with dev version of the library. Examples from Github dev branch works. The dev branch is working without problem. I'm looking forward to using release 1.0.