hybridgroup/gobot #346: bb8 driver missing NewDriver function

nickryand, 2016-12-23:
Trying to build on OSX and I get the following error while building:

```
$ GODBUG=cgocheck=0 go build -o /tmp/bb8 examples/bb8.go
# command-line-arguments
examples/bb8.go:14: undefined: bb8.NewDriver
$
```

Is NewDriver missing from the bb8_driver or am I doing something wrong? 

Attempting to build on:
```
$ go version
go version go1.7.3 darwin/amd64
```

I am building from the latest available via the `go get` instructions.

deadprogram, 2016-12-23:
Hi, @nickryand thanks for the report. Some code was indeed missing, and has been updated. You can now get the package with the `-u` option and the updated code is available.

Please reopen if needed, and thanks again.

closed 2016-12-23 by deadprogram

nickryand, 2016-12-23:
Working good now! Thanks for the quick response!