hybridgroup/gobot #348: Sphero BB-8 sample does not work on MacOS

durgeshm, 2016-12-28:
Referring to the "color change" sample here https://gobot.io/documentation/platforms/bb8/

BB-8 changes colors about 2 or 3 times and then stops. There are some errors as shown below, when the program starts.

$ GODEBUG=cgocheck=0 go run main.go BB-XXXX
2016/12/28 01:10:33 Initializing connections...
2016/12/28 01:10:33 Initializing connection BLECLient ...
2016/12/28 01:10:33 Initializing devices...
2016/12/28 01:10:33 Initializing device BB-8 ...
2016/12/28 01:10:33 Robot bb8 initialized.
2016/12/28 01:10:33 Starting Robot bb8 ...
2016/12/28 01:10:33 Starting connections...
2016/12/28 01:10:33 Starting connection BLECLient...
State: PoweredOn
scanning...
2016/12/28 01:10:37 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0xa7, 0xc3, 0x11, 0x24, 0x17, 0x9d, 0x46, 0x68, 0x8b, 0x61, 0x7a, 0xec, 0x8c, 0x76, 0xf9, 0x65}, "kCBMsgArgATTMTU":23}}

Connected Peripheral ID:a7c31124179d46688b617aec8c76f965, NAME:()
Failed to set MTU, err: Not implemented
2016/12/28 01:10:37 Unhandled event: xpc.Dict{"kCBMsgId":81, "kCBMsgArgs":xpc.Dict{"kCBMsgArgConnectionInterval":18, "kCBMsgArgConnectionLatency":0, "kCBMsgArgSupervisionTimeout":175, "kCBMsgArgDeviceUUID":xpc.UUID{0xa7, 0xc3, 0x11, 0x24, 0x17, 0x9d, 0x46, 0x68, 0x8b, 0x61, 0x7a, 0xec, 0x8c, 0x76, 0xf9, 0x65}}}
2016/12/28 01:10:39 Starting devices...
2016/12/28 01:10:39 Starting device BB-8...
2016/12/28 01:10:39 Starting work...

Same program when cross-compiled for ARM, runs fine on Linux (on a Raspberry Pi) - the colors keep changing indefinitely.

Also, tested the Cylon.js example color/color.js on the same Mac against the same BB-8. That also works correctly - colors changing indefinitely.

Thus, it looks like a problem specific to MacOS and Gobot.

$ go version
go version go1.7.4 darwin/amd64

deadprogram, 2016-12-28:
What version of OSX are you running?

durgeshm, 2016-12-28:
macOS Sierra  10.12.2

deadprogram, 2016-12-28:
Hi @durgeshm if you can, please give this branch a try https://github.com/hybridgroup/gobot/tree/feature/currantlabs-ble

It uses https://github.com/currantlabs/ble instead of https://github.com/currantlabs/gatt and seems to work well on my OSX with El Capitan.

durgeshm, 2016-12-28:
Yes, feature/currantlabs-ble branch works correctly. No errors. Thank you.

closed 2016-12-28 by durgeshm

mtoto, 2018-12-11:
I encountered the same issue on OSX High Sierra 10.13.6.

nehagup, 2019-11-23:
I encountered the same error on OSX Mojave 10.14.6

deadprogram, 2019-11-24:
Please see https://github.com/hybridgroup/gobot/issues/661 for more discussion on this topic.