hybridgroup/gobot #349: Sphero BB-8 driver - does it support collision detection?

durgeshm, 2016-12-29:
Not sure if I am missing something obvious, but does BB8Driver support collision detection?

I see that sphero.SpheroDriver has support for collision events. But BB8Driver does not use SpheroDriver at all. It basically uses ollie.Driver which does not seem to handle collisions as well. At least, that is how I am reading the code. Please correct me if I am missing anything.

deadprogram, 2016-12-29:
Hi @durgeshm you are correct, the implementation is still incomplete. There is a bunch of code in /platforms/sphero that can be reused for ollie and bb8, but I've not had a chance to work on that yet. Specifically all the packet creation code can be reused, since ollie & bb8 are the same API, but sent/received using BLE characteristics instead of Bluetooth serial.

A PR that worked on this would be gratefully accepted :)

durgeshm, 2016-12-29:
Thanks Ron. Just wanted to clarify if the code from sphero could be applied to ollie/bb8 except for BLE/serial and there wasn't some other fundamental difference. Thanks for confirming. I will see what I can do.

deadprogram, 2017-01-15:
Hi @durgeshm not sure if you are working on this, or possibly planning to, but you should check out the changes in the `dev` branch. We're now using https://github.com/currantlabs/ble for our BLE communication, and it works a lot better. There have been a couple internal API changes to be aware of that you can see in that branch.

durgeshm, 2017-01-15:
Is this different than the changes in master branch? It looks like 1.1 release already uses currantlabs/ble. Please correct me if I am missing something.

deadprogram, 2017-01-15:
https://github.com/hybridgroup/gobot/tree/dev is the dev branch.

1.1 uses currantlabs/gatt, but not currantlabs/ble which is pretty similar name.

durgeshm, 2017-01-16:
Is there a way to test the HandleResponses method? https://github.com/hybridgroup/gobot/blob/dev/platforms/sphero/ollie/ollie_driver.go#L154

It gets added as an event handler/subscriber at https://github.com/hybridgroup/gobot/blob/dev/platforms/sphero/ollie/ollie_driver.go#L107 . 

I also copied (and invoked) the ConfigureCollisionDetection method from sphero_driver. 

Under what conditions/events, is the "HandleResponses" method supposed to get called? Any insights would be great. Thanks.

deadprogram, 2017-01-17:
Hi @durgeshm the HandleResponses method is called whenever there is a data notification that comes in from the `responseCharacteristic`. When using any of the Sphero robots, you send commands to the `commandsCharacteristic` and then get the responses via the `responseCharacteristic`.

One way to test, would be to call HandleResponses directly using some known response byte arrays, and insure that whatever event handlers you are creating in your new implementation are being called as expected.

durgeshm, 2017-01-24:
Hi Ron,

The accompanied pull request (https://github.com/hybridgroup/gobot/pull/360) works for me. I have also added an example that is tested for BB-8.

Admittedly, I am new to BLE, but it was was not very obvious to me why ReadCharacteristic had to be called (repeatedly) in order to start receiving notification events (that were registered with Subscribe). But, this was the only way, HandleResponses started getting invoked with collision response packets.

deadprogram, 2017-01-25:
Hi @durgeshm the Sphero API does not start sending the collision packets until it receives the command packet to begin doing so. Then the notifications are just the means for sending the data. This is how the Sphero API works, which is an artifact of the serial interface on which the API was originally designed. I agree it is not exactly what a "normal" BLE app would do, but it certainly works once you understand the interaction model.

Anyhow, this is now in the `dev` branch, I will close the issue once the next release comes out. Thanks again!

deadprogram, 2017-02-16:
This has now been released as part of v1.2.0 thanks again!

closed 2017-02-16 by deadprogram