hybridgroup/gobot #354: undefined: syscall.SYS_IOCTL (build fails on Windows)

DerekL, 2017-01-06:
Looks like missing syscall functions, can't build on Windows. Looks like this: 

`C:\Work\pitest>go build -o test test.go
# gobot.io/x/gobot/sysfs
..\src\gobot.io\x\gobot\sysfs\i2c_device.go:60: undefined: syscall.SYS_IOCTL
..\src\gobot.io\x\gobot\sysfs\i2c_device.go:74: undefined: syscall.SYS_IOCTL
..\src\gobot.io\x\gobot\sysfs\i2c_device.go:107: undefined: syscall.SYS_IOCTL
..\src\gobot.io\x\gobot\sysfs\i2c_device.go:146: undefined: syscall.SYS_IOCTL
..\src\gobot.io\x\gobot\sysfs\syscall.go:32: not enough arguments in call to syscall.Syscall`

deadprogram, 2017-01-06:
Hi @DerekL you cannot build Gobot code intended for Raspi without using the correct cross-compilation flags as mentioned here https://gobot.io/documentation/platforms/raspi/#how-to-connect

The correct syntax in your case should be `GOARM=6 GOARCH=arm GOOS=linux go build -o test test.go`

DerekL, 2017-01-06:
Can confirm this works, thanks!

deadprogram, 2017-01-10:
Glad to have been of assistance. OK, closing.

closed 2017-01-10 by deadprogram