hybridgroup/gobot #359: ServoDriver / ServoWriter inconsistencies

erkkah, 2017-01-23:
While adding a ServoWriter implementation to c.h.i.p, I noticed that the ServoDriver accepts movement in angles (0-180) and calls the underlying ServoWriter with a span (0-255).

However, the ServoWriter implementations for raspi and beaglebone both expect movement argument to be in angles, which breaks when someone uses a ServoDriver to talk to the device.

Not sure how to proceed, and not cause more problems / break compatibility.

deadprogram, 2017-01-24:
Hi @erkkah that was an error that was introduced thru some incomplete refactoring. Thanks for the catch.

What we want to be happening, is that the `gpio.ServoDriver` should be passing along the angle value to `adaptor.ServoWrite()`. Then the specific adaptor can handle it however it needs to. In the Beaglebone case, by accepting an angle and scaling to a pwm duty cycle and calling PWMWrite. In the Firmata case, by just passing along the angle value.

So the conversion logic should be in the adaptor, not the `ServoDriver`.

erkkah, 2017-01-24:
Great, thanks, that makes sense.
Then I know and will adapt the implementation accordingly.

deadprogram, 2017-01-25:
I've just corrected the servo part with this commit https://github.com/hybridgroup/gobot/commit/53ffbbd2bb0d8671362480fc0e62badf43d2ac1b which is now in the `dev` branch so you should be all good for your C.H.I.P. implementation.

Thanks again @erkkah for the report.

closed 2017-01-30 by erkkah