hybridgroup/gobot #368: Re-authorize Coveralls

Zanadar, 2017-02-15:
Hey there!

I was looking through the project to see if I might add test coverage, and I saw the following in coveralls, when I clicked through to look at the specific coverage of a file:

[image]

deadprogram, 2017-02-16:
Thanks for reporting this @Zanadar seems like a coveralls problem, which we have reported.

I'll update this issue once it is resolved.

deadprogram, 2017-02-16:
Seems to be same problem as https://github.com/lemurheavy/coveralls-public/issues/596#issuecomment-138922490

deadprogram, 2017-02-16:
@Zanadar doing some experimenting with codecov.io check out https://codecov.io/gh/hybridgroup/gobot see what you think...

deadprogram, 2017-02-16:
OK, we've just "officially" switched to codecov.io :)

Closing this issue, please re-open if needed.

closed 2017-02-16 by deadprogram