hybridgroup/gobot #370: Platform request: BACnet/IP

dnishimura, 2017-02-16:
Any interest in having a BACnet/IP integration? It's mostly used in large industrial installations and integrates with most building management systems. I integrated BACnet/IP on the C/C++ side of things, but I'm curious to see if there are any interests here. It's a big undertaking since the BACnet specification is 1000+ pages, so I'd probably just do a subset of the implementation in Go.

deadprogram, 2017-02-16:
For sure that seems interesting, and also a daunting undertaking.

gen2thomas, 2024-02-21:
It seems nobody can spend the amount of work to implement the protocol for the [BACnet standard](https://bacnet.org/) since 7 years now. So I'm going to close this issue. Feel free to reopen, if you still have interest and can spend some time to start with a basic implementation.

closed 2024-02-21 by gen2thomas