hybridgroup/gobot #371: UART Support

kevinjos, 2017-02-19:
Has thought been put into what gobot/drivers/uart/uart.go might look like? I have a UART sensor that I am considering to control with gobot.

Examples of prior work on go serial drivers include:
https://godoc.org/github.com/pkg/term
https://godoc.org/github.com/tarm/serial

deadprogram, 2017-02-21:
Hi @kevinjos 

Which sensor are you thinking about?

We have a couple of platforms that already use serial connections. For example, Firmata (https://github.com/hybridgroup/gobot/blob/master/platforms/firmata/firmata_adaptor.go#L9), and also Sphero (https://github.com/hybridgroup/gobot/blob/master/platforms/sphero/sphero_adaptor.go#L8).

How we look at the architecture, is that an Adaptor that is part of a Gobot platform typically provide an interface that uses a platform-specific implementation. An example would be GPIO. Then Gobot drivers consume that interface.

Serial interfaces, on the other hand, are more like a transport than a protocol. The protocol is dependent on the device being communicated with. For example, a GPS sensor is not going to use a different serial interface, just a different port.

Anyhow, I hope that makes some sense. I'd like to know more about what you want to do, to better understand how to address your desired use case.

Thanks!

kevinjos, 2017-02-21:
Hi @deadprogram 

Here are the serial devices that I am hooking up to a BeagleBone Black:

http://www.co2meter.com/products/k-30-co2-sensor-module
https://learn.sparkfun.com/tutorials/openscale-applications-and-hookup-guide

I agree that serial UART is a transport for sensor specific protocol. I am thinking about serial UART as similar to I2C and GPIO.. a ReadWriteCloser with transport specific configurations (baud rate, data bits, parity bits, stop bits, etc.).

deadprogram, 2017-05-13:
Hi, @kevinjos 

I just switched several of the Gobot platforms that use serial communication over to use https://github.com/bugst/go-serial so this issue has returned to my attentions.

One thought, is to create an Adaptor to wrap around the serial communication, and then provide Drivers as needed. Those Drivers would not need necessarily to live inside of gobot/drivers/uart or gobot/drivers/serial or whatever one might wish to call it, although that also seems like a good place to put them.

Anyhow, what do you think?

kevinjos, 2017-05-15:
Hi @deadprogram,

Thanks for bumping this back up.

I like your idea to wrap github.com/bugst/go-serial and standardize interacting with serial devices on the Adaptor and Driver interface.

gen2thomas, 2024-11-05:
Fixed with [v2.4.0](https://github.com/hybridgroup/gobot/releases/tag/v2.4.0)

closed 2024-11-05 by gen2thomas