hybridgroup/gobot #376: Gobot hangs while finalizing connections

matipan, 2017-02-22:
Hi there!
This happened to me wihle running [a simple go program](https://gist.github.com/matipan/af6f0c5a4fa826984267e070d62027bd) :
```
2017/02/22 11:01:13 Initializing connections...
2017/02/22 11:01:13 Initializing connection Firmata-72720CD878A08339 ...
2017/02/22 11:01:13 Initializing devices...
2017/02/22 11:01:13 Initializing device LED-3FA3E55721925098 ...
2017/02/22 11:01:13 Robot bot initialized.
2017/02/22 11:01:13 Starting Robot bot ...
2017/02/22 11:01:13 Starting connections...
2017/02/22 11:01:13 Starting connection Firmata-72720CD878A08339 on port /dev/tty.usbmodem1411...
2017/02/22 11:01:18 Starting devices...
2017/02/22 11:01:18 Starting device LED-3FA3E55721925098 on pin 13...
2017/02/22 11:01:18 Starting work...

2017/02/22 11:01:21 HELLO

2017/02/22 11:01:22 HELLO
^C2017/02/22 11:01:24 Stopping Robot bot ...

2017/02/22 11:01:25 HELLO

2017/02/22 11:01:25 HELLO

2017/02/22 11:01:25 HELLO
2017/02/22 11:01:36 HELLO
2017/02/22 11:01:36 HELLO
2017/02/22 11:01:37 HELLO
2017/02/22 11:01:38 HELLO
2017/02/22 11:01:39 HELLO
^C^C^C^C
2017/02/22 11:01:45 HELLO
```
The go program never stops(even when I repeatedly send the interruption signal), It keeps executing what is inside the work func.

I think the problem is inside the [Stop](https://github.com/hybridgroup/gobot/blob/master/robot.go#L196) function, when it tries to finalize all the connections, for some reason it never returns from that method.

Thanks!
I'm using an Arduino UNO, although I don't think that is relevant

deadprogram, 2017-04-12:
The buffered channel code that seems to have addressed this has now been released as part of Gobot 1.4 so I am now closing this issue. Please reopen if needed. Thank you!

anweiss, 2017-04-28:
I'm currently seeing this issue using the latest release of Gobot. Running go 1.8.1 on macOS 10.12.4. Running simple program:

```go
package main

import (
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/firmata"
)

func main() {
	firmataAdapter := firmata.NewAdaptor("/dev/tty.usbmodem1421")
	led := gpio.NewLedDriver(firmataAdapter, "8")

	work := func() {
		gobot.Every(1*time.Second, func() {
			led.Toggle()
		})
	}

	robot := gobot.NewRobot("bot",
		[]gobot.Connection{firmataAdapter},
		[]gobot.Device{led},
		work,
	)

	robot.Start()
}
```

deadprogram, 2017-05-13:
Hello, everyone

I'm made some changes to the https://github.com/hybridgroup/gobot/tree/dev branch to refactor the Firmata client/adaptor as well as switching to https://github.com/bugst/go-serial for all serial port communication.

This bug appears to be fixed from my testing on OSX El Capitan. If anyone else can confirm or deny that it is  working for them, it would be greatly appreciated.

Thanks!

deadprogram, 2017-06-15:
Version 1.6.0 is now out, and includes fixes for this. Now closing, please reopen if needed. Thank you!

closed 2017-06-15 by deadprogram