hybridgroup/gobot #387: gobot fails with go1.5

rgdaddio, 2017-03-13:
go version go1.5 linux/amd64

go get  -d -u gobot.io/x/gobot/...

package gobot.io/x/gobot/examples
	imports context: unrecognized import path "context"

deadprogram, 2017-03-15:
Hi, @rgdaddio we dropped official support for versions of Golang < 1.7 in the last release. I just realized that we did not add that important information to the CHANGELOG, and I apologize for that.

Do you need to use golang v1.5 for a particular reason?

rgdaddio, 2017-03-15:
Ok thanks.

I have migrated to golang v1.7 but it should be noted that the migration to < 1.7 does not work with gccgo for current official releases of GCC.

Users wishing to use gccgo for gobot.io will have to migrate to GCC-7.0 which currently supports golang v1.8 --note GCC-7.0 is not officially released at this time. 

AFAICT there is no gccgo version that supports anything between golang v1.5 and v1.8. 

svn checkout svn://gcc.gnu.org/svn/gcc/trunk .

gen2thomas, 2022-10-19:
Meanwhile the current golang version is [1.19](https://go.dev/doc/install), so I will close this issue.

closed 2022-10-19 by gen2thomas