hybridgroup/gobot #390: gobot.io will not build with gcc7.0 (v1.8)

rgdaddio, 2017-03-21:
I have tested gobot.io Arduino example with GCC-7.0 (gccgo)and golang v1.8 and there are build errors due to the following known go issue: 
https://github.com/golang/go/issues/15628

There are workarounds if gcc is required. 

Builds fine with the standard compiler.

trex58, 2017-03-27:
Would you mind elaborate about how a workaround can be built ?

rgdaddio, 2017-03-27:
At least in the Gobot with Arduino case I tested, the file causing the build error is: 

github.com/hashicorp/go-multierror/prefix.go

I don't believe this file is actually used with this architecture as the link completes without it. And so can probably be safely removed or commented out to build and run successfully. 

I built and ran with GCC-7.0 and golang v1.8 with the gccgo option without the prefix.go file and had no issues in a very basic compile and run it test.

HTH, 
Rich

trex58, 2017-03-28:
Thx.
However, I expected a generic patch, for any Go product with the 15628 issue.
I'll reread 15628 explanations.
Tony

rgdaddio, 2017-03-28:
It didn't seem like there were anything other than "dodgy" workarounds for issue in 15628 AFAICT. Not aware of any patches. This issue with gccgo has been around for almost a year. 

The best current workaround for our project was to use the standard gc compiler which works fine and seems to have much more consistent support than gccgo at this point. 

gccgo is not supported for golang v1.6 and golang v1.7. in the GCC-6.x releases (golang v1.5) series. Maybe there are patches I don't know. 

GCC-7.0 does support golang v1.8. So that means that you cannot use GCC with current gobot.io unless you use the yet unreleased GCC-7.0,  since gobot.io no longer support anything below golang v1.7: https://github.com/hybridgroup/gobot/issues/387

HTH,

Rich

gen2thomas, 2022-10-19:
Seems to be closed and/or outdated., so I will going to close this issue.

closed 2022-10-19 by gen2thomas