hybridgroup/gobot #391: Golang on Rasberry Pi

SafaiWeb, 2017-03-22:
Do I need to install the drivers on my pi, or is compiling on my dev computer and transferring sufficient?

The Getting Started guide is a little ambiguous in this regard.

trevrosen, 2017-04-07:
@SafaiWeb you can compile on your local machine as long as you're not using any cgo stuff.

Something like: `GOOS=linux GOOSARCH=arm7 go build`

Then you can do like `file foo` and see that `foo` has been built as an `ELF` for ARM.

Hope that helps!

deadprogram, 2017-04-12:
Hello @SafaiWeb & @trevrosen please take a look at my changes to the README here https://github.com/hybridgroup/gobot/commit/95a433a84aff04a13698950448f8816ee2da6c06

I think they should make things a bit clearer.

deadprogram, 2017-04-12:
These docs have now been released as part of Gobot 1.4 so I am now closing this issue. Please reopen if needed. Thank you!

closed 2017-04-12 by deadprogram