hybridgroup/gobot #398: Gobot Bebop: Fatal Error Concurrent Map Writes

mccurdyc, 2017-03-29:
When trying to use the API and Bebop libraries, I get the following issue. Before trying this, I was just using http handler functions and received the same issue. I have also posted a minimal example. The commands DO work. I am able to takeoff and land, then the program crashes.

```
2017/03/29 14:50:17 Initializing API on :3000...
2017/03/29 14:50:17 Initializing connections...
2017/03/29 14:50:17 Initializing connection Bebop-361A1C107A69ECB7 ...
2017/03/29 14:50:17 Initializing devices...
2017/03/29 14:50:17 Initializing device Bebop-32188FCF05C4CB81 ...
2017/03/29 14:50:17 Robot drone initialized.
2017/03/29 14:50:17 Starting Robot drone ...
2017/03/29 14:50:17 Starting connections...
2017/03/29 14:50:17 Starting connection Bebop-361A1C107A69ECB7...
2017/03/29 14:50:17 WARNING: API using insecure connection. We recommend using an SSL certificate with Gobot.
2017/03/29 14:50:17 Starting devices...
2017/03/29 14:50:17 Starting device Bebop-32188FCF05C4CB81...
2017/03/29 14:50:17 Starting work...
fatal error: concurrent map writes

goroutine 11 [running]:
runtime.throw(0x7030e4, 0x15)
	/usr/lib/go/src/runtime/panic.go:596 +0x95 fp=0xc42003abf0 sp=0xc42003abd0
runtime.mapassign(0x6a99a0, 0xc420015590, 0xc42003acd9, 0xc4200b8af3)
	/usr/lib/go/src/runtime/hashmap.go:499 +0x667 fp=0xc42003ac90 sp=0xc42003abf0
gobot.io/x/gobot/platforms/parrot/bebop/client.networkFrameGenerator.func1(0xc42015a770, 0xc420150d02, 0x8)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/platforms/parrot/bebop/client/client.go:127 +0xd9 fp=0xc42003ad20 sp=0xc42003ac90
gobot.io/x/gobot/platforms/parrot/bebop/client.(*Bebop).createARStreamACK(0xc4200d6000, 0x1c7, 0x0, 0x0, 0x1, 0xc42014a00c, 0x1971, 0x9ff4, 0x1971)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/platforms/parrot/bebop/client/client.go:795 +0x703 fp=0xc42003ae18 sp=0xc42003ad20
gobot.io/x/gobot/platforms/parrot/bebop/client.(*Bebop).packetReceiver(0xc4200d6000, 0xc42014a000, 0x197d, 0xa000)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/platforms/parrot/bebop/client/client.go:546 +0x227 fp=0xc42003af30 sp=0xc42003ae18
gobot.io/x/gobot/platforms/parrot/bebop/client.(*Bebop).Connect.func2(0xc4200d6000)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/platforms/parrot/bebop/client/client.go:290 +0x19f fp=0xc42003afd8 sp=0xc42003af30
runtime.goexit()
	/usr/lib/go/src/runtime/asm_amd64.s:2197 +0x1 fp=0xc42003afe0 sp=0xc42003afd8
created by gobot.io/x/gobot/platforms/parrot/bebop/client.(*Bebop).Connect
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/platforms/parrot/bebop/client/client.go:292 +0x322

goroutine 1 [chan receive]:
gobot.io/x/gobot.(*Master).Start(0xc420016380, 0x6fe265, 0x4)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/master.go:70 +0x118
main.main()
	/home/mccurdyc/allegheny/cs383s2017/cs383s2017-team1-lab6/main.go:47 +0x428

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/usr/lib/go/src/runtime/asm_amd64.s:2197 +0x1

goroutine 5 [syscall]:
os/signal.signal_recv(0x0)
	/usr/lib/go/src/runtime/sigqueue.go:116 +0x104
os/signal.loop()
	/usr/lib/go/src/os/signal/signal_unix.go:22 +0x22
created by os/signal.init.1
	/usr/lib/go/src/os/signal/signal_unix.go:28 +0x41

goroutine 6 [chan receive]:
gobot.io/x/gobot.NewEventer.func1(0xc4200b86a0)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/eventer.go:65 +0x6c
created by gobot.io/x/gobot.NewEventer
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/eventer.go:73 +0x154

goroutine 7 [chan receive]:
gobot.io/x/gobot.NewEventer.func1(0xc4200b86c0)
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/eventer.go:65 +0x6c
created by gobot.io/x/gobot.NewEventer
	/home/mccurdyc/dev/gocode/src/gobot.io/x/gobot/eventer.go:73 +0x154

goroutine 8 [IO wait]:
net.runtime_pollWait(0x7fbda2695f00, 0x72, 0x0)
	/usr/lib/go/src/runtime/netpoll.go:164 +0x59
net.(*pollDesc).wait(0xc420110068, 0x72, 0x0, 0xc420232000)
[... 118 lines cut ...]
		return "Master says hello!"
	})

	d := gbot.AddRobot(gobot.NewRobot("drone",
		[]gobot.Connection{bebopAdaptor},
		[]gobot.Device{drone},
	))

	// Accessible via http://localhost:3000/robots/Eve/commands/say_hello
	d.AddCommand("say_hello", func(params map[string]interface{}) interface{} {
		return fmt.Sprintf("%v says hello!", d.Name)
	})

	d.AddCommand("takeoff", func(params map[string]interface{}) interface{} {
		drone.TakeOff()
		return 0
	})

	d.AddCommand("land", func(params map[string]interface{}) interface{} {
		drone.Land()
		return 0
	})
	gbot.Start()
}
```

mccurdyc, 2017-03-29:
Here is information about my Go version: `go version go1.8 linux/amd64`
And system information: `4.9.11-1-ARCH x86_64 GNU/Linux`
Wireless chip: `04:00.0 Network controller: Intel Corporation Wireless 8260 (rev 3a)`

Let me know if there is any additional information that would help you! Thank you in advance!

losinggeneration, 2017-07-15:
@mccurdyc After GopherCon would you be able to give my branch a try?

mccurdyc, 2017-07-15:
@losinggeneration I would love to, but I don't have access to the drone anymore. It was a course I was taking in school.

losinggeneration, 2017-07-16:
I talked with @deadprogram and he said he has some, it's just a matter of digging them out and trying it out.

matiasinsaurralde, 2017-08-31:
I'm having the same issue, also I'm not able to get the video when using Bebop 2, any idea?

gen2thomas, 2024-11-05:
Fixed with [v2.4.0](https://github.com/hybridgroup/gobot/releases/tag/v2.4.0)

PR #1063 (2024-02-12): bebop: fix concurrent map writes
## Solved issues and/or description of the change

Fix #398 by introducing a dedicated network frame generator object and mutex on generate.

## Manual test

none

## Checklist

- [x] The PR's target branch is 'hybridgroup:dev'
- [x] New and existing unit tests pass locally with my changes (e.g. by run `make test_race`)
- [x] No linter errors exist locally (e.g. by run `make fmt_check`)
- [x] I have performed a self-review of my own code
closed 2024-11-05 by gen2thomas