hybridgroup/gobot #399: Raspberry Pi 3 Completely Unresponsive

SafaiWeb, 2017-03-29:
I'm sorry if there is something I am not seeing in here, I'm really trying to use this framework and any help would be appreciated:

I have had issues with this for days using a RP3 and the newest version of Raspbian.

I copy/pasted the program from the platform page at https://gobot.io/documentation/platforms/raspi/.

I set up the circuit exactly as shown in the led light diagram.

The program compiled without errors (cross-compiled with GOARM=7 GOARCH=arm GOOS=linux).

When I run the program from the Pi, I get no response.

I tested the circuit against one of the power pins, and the LED lights up.

But, when used with the GPIO pins, I can't get so much as a flicker.  

Any suggestions or am I just going crazy?

deadprogram, 2017-03-30:
Hello, @SafaiWeb perhaps you can upload a photo of your exact setup, in particular your circuit.

SafaiWeb, 2017-03-30:
Thank you so much, I feel like it has to be something very silly on my part:

[image]
[image]
[image]

deadprogram, 2017-03-30:
Looks like the LED is reversed in that photo. The flat side should be GND (negative). Normally I use a black wire to indicate ground, and use colors for pins. I reserve red for power.

Hope that helps!

SafaiWeb, 2017-03-30:
I have tried it in both directions just to be sure :(

deadprogram, 2017-03-30:
It will only work in one direction, that is for sure! :) Flat means GND.

deadprogram, 2017-03-30:
Also, looks like you are using a 1K ohm resistor. You might want to try 270 ohm instead.

SafaiWeb, 2017-04-01:
Have now tried a different resistor.  Have tried a different RP3.  No response.

Took circuit and switched to an arduino UNO. Followed instructions, used Gort. Uploaded Firmata, copy-pasted led blinking program.  No response.  The only discernable difference between the arduino and the RP3 is that when I execute the compiled program for the RP3 the terminal says "starting work..." before not responding, whereas with the arduino it says "starting connection firmata...on port...." before giving me a new command prompt and also not responding.

I'm as lost as a lamb in a snowstorm as to what could be wrong.

SafaiWeb, 2017-04-01:
I also attached a PIR Motion Sensor to both the PI and the Arduino with the hopes of printing a line on the screen when motion was detected.  The Pi gave no response to either motion or lack of motion.

The Arduino returns a new command prompt, so there was no way to print to the terminal.  

Given the strange behavior of lights, and inability of motion sensing technology to detect me, current working theory is that I am a vampire.

deadprogram, 2017-04-01:
Hi @SafaiWeb I'm not exactly sure what you mean by "The Arduino returns a new command prompt, so there was no way to print to the terminal."

Perhaps bringing in both an Arduino and other sensors into the mix might be clouding your original issue, as opposed to clearing it. Not sure on that, perhaps it is helping establish a baseline.

In any case, a few more basic questions:
- What version of OS X are you using on your desktop machine?
- What version of Golang do you have installed?
- How exactly did you load Fimata onto your Arduino?
- Which Arduino model are you using?
- What is the command line you are using to run the firmata_blink.go program?

SafaiWeb, 2017-04-01:
I wasn't sure if it was helpful to change up the hardware or not. I thought there might be a potential defect in my actual pi, but it seems my issue is not there. 

1. MacOS Sierra. I have also tried using an Ubuntu 14.04 installation on a separate computer. 
2. I am using Golang 1.8. Installed via the binary on the Go project webpage on the mac. Installed via gvm on tge ubuntu machine
3. I uploaded with gort using the "gort arduino upload firmata /dev/ttyACM0". As this is the default address I also tried it once without the /dev/ttyACM0. This was the address given by gort scan serial. I really like Gort, by the way. 
4. Arduino UNO R3. 
5. Go run firmata_blink.go

deadprogram, 2017-04-01:
If you are running the firmata_blink.go from your OSX machine to the connected Arduino, you would need to change up the serial interface name here https://github.com/hybridgroup/gobot/blob/master/examples/firmata_blink.go#L16 to match the name of whatever serial port your Arduino is mapped to on OS X.

SafaiWeb, 2017-04-01:
I am running it on my Ubuntu machine, gort scan serial gives me the address [/dev/ttyACM0] as the address for my arduino.

deadprogram, 2017-04-01:
Ubuntu happens to be my own preferred desktop OS. Perhaps permissions related? Try building, and then execute your program using sudo.

deadprogram, 2017-04-01:
You probably will need to run your code on the raspi using sudo as well, unless you have added udev rules to give yourself whatever needed permissions to read/write to GPIO.

deadprogram, 2017-04-01:
OK so I have a solution now thanks to a tip from @fivdi to ensure that the GPIO exports have correctly happened before trying to use the GPIO pins. This addresses the issue when running as a normal user.

The code is now in the `dev` branch @SafaiWeb if you want to try it out on your raspi. I will leave this issue open until this code has made it into the next release.

Thanks you @SafaiWeb for pointing this problem out.

trevrosen, 2017-04-07:
@deadprogram just chiming in to say I think this export check is a great idea and something that has bitten me in the past, so I'm excited to see it land on mainline.

deadprogram, 2017-04-12:
This code has now been released as part of Gobot 1.4 so I am now closing this issue. Please reopen if needed. Thank you everyone!

closed 2017-04-12 by deadprogram