hybridgroup/gobot #401: importing firmata causes a `signal: killed`

matipan, 2017-04-01:
Hi there, I have an issue that appeared a few days ago. The import `gobot.io/x/gobot/platforms/firmata` causes a `signal: killed` when trying to run a simple program.

This is how I verified it was that import:
```go
package main

import (
	"log"

	_ "gobot.io/x/gobot"
	_ "gobot.io/x/gobot/drivers/gpio"
	_ "gobot.io/x/gobot/platforms/firmata"
)

func main() {
	log.Println("hello")
}
```
When you do a `go run main.go` to that file, it prints a `signal: killed`  
Now, if we run this program:
```go
package main

import (
	"log"

	_ "gobot.io/x/gobot"
	_ "gobot.io/x/gobot/drivers/gpio"
	// _ "gobot.io/x/gobot/platforms/firmata"
)

func main() {
	log.Println("hello")
}
```
It logs as I expected `hello`.

So there seems to be an issue with the `firmata` import. Any idea what it could be?

go version:`go version go1.8 darwin/amd64`
I'm on macOS

deadprogram, 2017-04-01:
Hmm, @matipan only thing I could find on this error is https://forum.golangbridge.org/t/go-build-exits-with-signal-killed/513

deadprogram, 2017-04-01:
OK seems like an open issue possibly https://github.com/golang/go/issues/19734

matipan, 2017-04-02:
Ok, thanks for pointing out that issue. It seems to be a problem with Xcode 8.3. I checked the update logs and the day this started to happen is the day Xcode got updated.

So, for anybody that reads this issue, a quick workaround is to build the program using the `-ldflags` and `-s` flags. There seems to be with an issue with how the linker in Darwin invokes dsymutil. The `-s` flag disables that invocation.

Another workaround is going back to Xcode 8.2

According to what is being discussed they will try to have it fix for go 1.8.1, but there are no guarantees.

As always, thanks for the quick response @deadprogram!

closed 2017-04-02 by matipan