hybridgroup/gobot #404: Microbit OSX signal:killed

willis7, 2017-04-07:
I can't seem to get the ubit to work with gobot. I did the firmware install with gort:

```
gort microbit download
gort microbit install /Volumes/MICROBIT
```

but when I run (with the example):

`GODEBUG=cgocheck=0 go run main.go "BBC micro:bit [tupov]"`

I get `signal: killed`.

I used LightBlue to discover the Bluetooth name on the mac, but the ubit isn't doing anything.

Happy to make a PR with doc updates if I can fix it.

deadprogram, 2017-04-08:
Hi @willis7 I think your problem is probably the same as https://github.com/hybridgroup/gobot/issues/401

willis7, 2017-04-09:
Thanks for getting back to me. If I add the flags as suggested then the application gets a little further, but there's still no output on the device:

```
wills7:pomodoro-bit/ $ GODEBUG=cgocheck=0 go run -ldflags -s main.go "BBC micro:bit [tupov]"                                                                                                                                                                         [17:05:00]
2017/04/09 17:05:07 Initializing connections...
2017/04/09 17:05:07 Initializing connection BLEClient-666C9345B9D5A4E4 ...
2017/04/09 17:05:07 Initializing devices...
2017/04/09 17:05:07 Initializing device Microbit LED-583534B99F62E730 ...
2017/04/09 17:05:07 Robot blinkBot initialized.
2017/04/09 17:05:07 Starting Robot blinkBot ...
2017/04/09 17:05:07 Starting connections...
2017/04/09 17:05:07 Starting connection BLEClient-666C9345B9D5A4E4...
2017/04/09 17:05:10 Starting devices...
2017/04/09 17:05:10 Starting device Microbit LED-583534B99F62E730...
2017/04/09 17:05:10 Starting work...
```

deadprogram, 2017-04-10:
Hi @willis7 try connecting by ID instead of name, that might help you out.

deadprogram, 2017-04-10:
This might sound silly, but did you rotate the Microbit in a circle first after flashing the firmware, but before trying to connect?

Step 4 from here:
https://github.com/sandeepmistry/node-bbc-microbit#flashing-microbit-firmware

We absolutely need to add this to the Gobot docs.

deadprogram, 2017-04-12:
The docs have now been updated. @willis7 did that solve your problem?

willis7, 2017-04-12:
Hi @deadprogram,

I stupidly took the ubit into work and haven't been able to test it. I will test it tomorrow and get back to you.

If it helps I had actually performed the calibration step. I haven't tried connecting by ID, so I shall give that a go.

deadprogram, 2017-04-27:
Hi @willis7 any update on testing this? Thanks!

willis7, 2017-05-02:
Apologies for the delay.

I had performed the rotation and got the smiley face.

I tried using the UUID of the device, but it didn't even reach the `Starting work...` line in the logs.

Are there any restrictions on versions? I'm fairly certain I had one of the pre-release devices.

deadprogram, 2017-05-03:
No restrictions I am aware of. That said, I do almost all of my work on Linux.

The `ble` package that Gobot uses is different for Linux vs. OSX, in that the OSX code uses the Mac's API. I will need to setup to do some further testing.

willis7, 2017-05-03:
Is there anything I can do my end to support the debugging?