hybridgroup/gobot #411: How to stop a robot

triberraar, 2017-04-22:
I am using gobot to lean go, so please don't judge my code too harshly.
I am using gobot togheter with a sphero and want do do some simple things, right now just flashing some colors. My code looks like this:
```
package main

import (
	"bufio"
	"fmt"
	"log"
	"os"
	"strings"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/platforms/sphero"
)

type color struct {
	r uint8
	g uint8
	b uint8
}

func main() {
	reader := bufio.NewReader(os.Stdin)
	fmt.Print("Enter the sphero port: ")
	spheroPort, _ := reader.ReadString('\n')
	// spheroPort = "/dev/tty.Sphero-GRY-AMP-SPP"
	spheroPort = strings.TrimSpace(spheroPort)

	adaptor := sphero.NewAdaptor(spheroPort)
	driver := sphero.NewSpheroDriver(adaptor)

	go goSphero(driver, adaptor)
}

func goSphero(driver *sphero.SpheroDriver, adaptor *sphero.Adaptor) {
	red := color{255, 0, 0}
	green := color{0, 255, 0}
	blue := color{0, 0, 255}
	yellow := color{255, 255, 0}
	work := func() {
		fmt.Print("Colors used are: ")
		fmt.Print("Red")
		driver.SetRGB(red.r, red.g, red.b)
		time.Sleep(1000 * time.Millisecond)
		fmt.Print("Green")
		driver.SetRGB(green.r, green.g, green.b)
		time.Sleep(1000 * time.Millisecond)
		fmt.Print("Blue")
		driver.SetRGB(blue.r, blue.g, blue.b)
		time.Sleep(1000 * time.Millisecond)
		fmt.Print("Yellow")
		driver.SetRGB(yellow.r, yellow.g, yellow.b)
		time.Sleep(1000 * time.Millisecond)
		// would like to disconnect from the robot or stop the program
	}

	robot := gobot.NewRobot("sphero",
		[]gobot.Connection{adaptor},
		[]gobot.Device{driver},
		work,
	)

	error := robot.Start()
	if error != nil {
		fmt.Print("something went horribly wrong")
		log.Fatal(error)
	}
}
```

Where i have the comment i would like the program to end or to disconnect from the robot. I tried inserting driver.Stop() at several places, but it didn't help me. Can you help me with what I am missing?

Eventually i would like to write a webservice where you can post a color and the sphero will flash in this color. I am a bit confused that you want a robot to do has to be in the work function. So any help with that is also appreciated.

deadprogram, 2017-04-28:
Hi @triberraar if what you want to do is immediately terminate your Gobot program after running, instead of continuing to run and await commands and events, you can pass `false` to the `Start()` method, which turns off the "autorun" feature.

For example:

```go
package main

import (
	"fmt"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/platforms/sphero"
)

func main() {
	adaptor := sphero.NewAdaptor("/dev/rfcomm0")
	spheroDriver := sphero.NewSpheroDriver(adaptor)

	work := func() {
		r := uint8(gobot.Rand(255))
		g := uint8(gobot.Rand(255))
		b := uint8(gobot.Rand(255))
		spheroDriver.SetRGB(r, g, b)
	}

	robot := gobot.NewRobot("sphero",
		[]gobot.Connection{adaptor},
		[]gobot.Device{spheroDriver},
		work,
	)

	robot.Start(false)
}
```

Notice that when the program disconnects from the Sphero, the Sphero will turn off once the Bluetooth connection is broken. This is the behavior of the Sphero's firmware, not something specific to Gobot.

Hope that helps!

gm42, 2019-05-27:
I would be interested in an example for graceful termination e.g. upon SIGTERM or SIGINT, expect the robot to terminate gracefully and exit.

I could not find an (exposed) channel or context to implement such pattern. I have read in another comment (#405) about this and I agree with those conclusions.

peteut, 2020-04-17:
If a robot is used with `systemd`, `KillSignal=SIGINT` must be set, as robot looks for `os.Interrupt`.