hybridgroup/gobot #416: GPIO Interrupt Support

pauleyj, 2017-05-05:
Any thoughts or plans on supporting GPIO interrupts?

larsth, 2017-05-12:
Epoll is your friend:

https://godoc.org/syscall#EpollCreate
https://godoc.org/syscall#EpollCreate1
https://godoc.org/syscall#EpollCtl
https://godoc.org/syscall#EpollWait

epoll(7) man page: http://man7.org/linux/man-pages/man7/epoll.7.html
epoll(2) man page: https://linux.die.net/man/4/epoll

Use EPOLLET mode / flag when using EpollCreate, EpollCreate1

mikegleasonjr, 2017-10-22:
So should we implement this in userland and not user specific drivers?

deadprogram, 2017-10-22:
Hi everyone. Somehow I previously missed @larsth comment when it was originally made. Yes, epoll would be a great solution. Seems like by adding some wrapper code to the Gobot sysfs package, getting to work on any Linux such as Raspi, Beaglebone Black, etc. would be straightforward.

@larsth if there is a kernel module that blocks the interrupt, that probably means a kernel driver that is already using it, and so it would be best for Gobot to leave it alone, right? What is the correct way to handle this?

@mikegleasonjr yes, that seems like the most likely pattern for a Gobot implementation. I personally avoid adding kernel modules unless absolutely needed for some special purpose.

larsth, 2017-10-23:
> @larsth if there is a kernel module that blocks the interrupt, that probably means a kernel driver that is already using it, and so it would be best for Gobot to leave it alone, right? 

Yes

> What is the correct way to handle this?

The only thing that can be done is to remove the module from the kernel:
`sudo rmmod >name of the offending kernel module goes here<`

What happens after that command shows whether or not removing the module from the kernel was a good idea, however, I think it would be safe to remove the module, because the user of Gobot is probably doing something else with the GPIO pins.

Controlling a piece of hardware with only one GPIO pin at a time is highly undesirable because context switches creates long delays. 
If only there was a GPIO kernel module with poll syscall support it would be great, but I don't have any time anytime soon to create such a module. The 2 times earlier that I had made a character device module was very fun. It was 9 years ago.

BTW,
It is fun to watch that after fx a shell command like this:
`echo 1 > /dev/my_character_device_file` 
... there continues to be action, because inertia prevents anything to happen instantly.

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas