hybridgroup/gobot #420: firmata: analog pin number to digital mapping is not always +14

1lann, 2017-05-14:
In `firamta_adaptor.go` line 221: https://github.com/hybridgroup/gobot/blob/51f6aba47c369d95e395fca722d91ea48037ecda/platforms/firmata/firmata_adaptor.go#L221

The digital mapping for analog pins is hard coded as `+ 14`, however, this is not always true for some boards. For example, on the Arduino Yún, it's `+ 18`. This should be documented and/or changed to allow people to specify offset, have a database of boards, or somehow detect the correct offset value.

deadprogram, 2017-11-16:
Hello @1lann was finally looking into this, and I see that actually the `Client` is doing a `AnalogMapping` query, and so obtains the actual pin mappings for any board on initial connection.

The problem is that the Adaptor is not using this information, instead using the "hard wired" value as you pointed out. I have not had a chance to actually work on the code, but https://github.com/hybridgroup/gobot/blob/master/platforms/firmata/client/client.go#L450 is where the values are populated. The https://github.com/hybridgroup/gobot/blob/master/platforms/firmata/firmata_adaptor.go#L239 function should probably be renamed to `analogPinLookup()` and the implementation use the f.Board.Pins() to lookup the actual mapping.

Hopefully this makes sense. I will try to get to working in the next weeks on an improvement to handle this case, but if someone else wanted to write a PR to handle this, it would be greatly appreciated!

kVinsom, 2026-10-31:
I'm working on this. I'll use the analog-channel mapping reported by Firmata instead of the fixed +14 offset, and add regression tests for non-Uno layouts, unmapped channels, and the pin/channel arguments sent to the board.