hybridgroup/gobot #421: Stuck at starting connection (with Firmata)

conejoninja, 2017-05-18:
I'm trying to connect to an Arduino Uno with Standard Firmata running on it, tried 3 different USB-cables and two different USB ports. Debian Jesie and user is on dialout group if that helps.

This issue happens _most of the time_, but not always. It just hangs at

```
➜  examples go run examplebot.go               
2017/05/18 09:39:17 Initializing connections...
2017/05/18 09:39:17 Initializing connection Firmata-5721E10BB1A30F02 ...
2017/05/18 09:39:17 Initializing devices...
2017/05/18 09:39:17 Initializing device LED-77F786BE130713F4 ...
2017/05/18 09:39:17 Robot sensorBot initialized.
2017/05/18 09:39:17 Starting Robot sensorBot ...
2017/05/18 09:39:17 Starting connections...
2017/05/18 09:39:17 Starting connection Firmata-5721E10BB1A30F02 on port /dev/ttyUSB0...
```

And that's it, it stays there, nothing more.

More information, I'm working on a _driver_ for a sensor, maybe the code crash or exits without closing properly the connection to /dev/ttyUSB0  (as an idea). But If I disconnect/re-connect the arduino, the problem is still there. Arduino IDE is able to flash the board without any problem, it only happens with gobot's code. Also, once it start happening, no matter what I run (my code, or gobot's examples), it's stuck there.

Probably it's a configuration issue of my setup, but not sure how to identify what is wrong

deadprogram, 2017-05-18:
The easy way to establish if it is permissions related, is to run your program under sudo.

If it runs correctly, it is permissions related. If not, it is something else.

Please give that a try.

Also, from where did you obtain your Firmata? Try using the Firmata that is uploaded using Gort (http://gort.io/) it was created using ConfigurableFirmata and includes all the needed capabilities.

conejoninja, 2017-05-18:
Same results with sudo and chmod the file, sometime it works, sometime it doesn't.
I tried with both, installing firmata from gort and from the IDE.

I will try with a different board, in case something is wrong with this one, since it works at times, it may be the problem. I'm running out of ideas.

deadprogram, 2017-05-18:
Did you try switching to the `dev` branch of Gobot? There are some important Firmata/serial related changes in that branch.

conejoninja, 2017-05-18:
Yes, I'm using dev branch, last commit 4e05c7f2c60604ae6f67ac4ae8f69378f6616d32

Even if this is a hardware problem, I think there could be some timeout, or at least some verbose option, that will help sort this out.

conejoninja, 2017-05-18:
I close this issue, I used a different board and so far so good, no problems for now. Looked like a hardware problem.

half2me, 2017-05-23:
I have the same problem with Arduino UNO and OSX 10.12.4
```
2017/05/23 15:51:22 Initializing connections...
2017/05/23 15:51:22 Initializing connection Firmata-A283223A5671633 ...
2017/05/23 15:51:22 Initializing devices...
2017/05/23 15:51:22 Initializing device LED-62505DE4731EE1DF ...
2017/05/23 15:51:22 Robot bot initialized.
2017/05/23 15:51:22 Starting Robot bot ...
2017/05/23 15:51:22 Starting connections...
2017/05/23 15:51:22 Starting connection Firmata-A283223A5671633 on port /dev/cu.usbmodem1421...
```

deadprogram, 2017-05-23:
Hi @half2me did you try the `dev` branch as I had suggested?

half2me, 2017-05-23:
@deadprogram I'm just starting to learn go, so I'm not sure how I would go about doing that :)

conejoninja, 2017-05-23:
@half2me go to your GOPATH/src/gobot.io/x/gobot/  and there just change the git branch with 

```bash
git checkout dev && git pull origin dev

```

deadprogram, 2017-05-24:
We will release an update with these fixes once we have confirmation from someone else that this issue and https://github.com/hybridgroup/gobot/issues/376 have been addressed in the `dev` branch. 

We've tested on OSX El Capitan, just need testing on some more recent OSX versions.

half2me, 2017-05-24:
@deadprogram  @conejoninja I just tested with dev branch... same problem :(

deadprogram, 2017-05-24:
@half2me did you flash the Arduino with Firmata first?

half2me, 2017-05-25:
@deadprogram nope :/ I didn't know I had to do that, didn't see it documented anywhere...

deadprogram, 2017-05-25:
https://gobot.io/documentation/platforms/arduino/ first paragraph in "How to install" documents this.

deadprogram, 2017-05-26:
@half2me I added some code to the `firmata.Connect()` to some ideas that have emerged from this issue. The new code is much cleaner, and provides a new `ConnectTimeout` field that defaults to 15 seconds.

In the case that Gobot is unable to connect, the updated Firmata "blink" example will display any connection errors: https://github.com/hybridgroup/gobot/blob/dev/examples/firmata_blink.go#L32

Also, the error displayed in the case of a timeout, suggests flashing Firmata on your Arduino, for people who might not realize that requirement.

I am going to reopen this issue until the new code makes it into a Gobot release. In the meantime, anyone who can test the `dev` branch, your feedback in appreciated.

conejoninja, 2017-05-26:
Nice, I can confirm the timeout is working properly, much better than before. In my case, it was a faulty hardware issue, that sometimes worked, sometimes not.

half2me, 2017-05-26:
@deadprogram Awesome work, thank you! I will test the dev branch as much as possible and report any bugs I might find. 👍

JDogHerman, 2017-06-14:
Seems to be working for me also, The dev branch let me know I was having a permissions error. After running ```ln -s /usr/local/go/bin/go /usr/bin/go``` I was able to run ```sudo go run blink.go``` without issue.

deadprogram, 2017-06-15:
Version 1.6.0 is now out, and includes all these fixes. Now closing, please reopen if needed. Thank you everyone!

closed 2017-06-15 by deadprogram