hybridgroup/gobot #422: kuka proxy

gedw99, 2017-05-19:
i knwo this is for embedded hardware but the principles for industrial and non fenced robotics apply in the same way.

Has anyone attempted using gobot with the low level kuka movement language ?
Its E6Pos and easy to codegen with golang.

The big advantage is that you can then use parametrics to drive it and so then have real time adjusting bots. The sensor streams tells you whats happens and the parametrics lib adjust the streaming output to the robot of the E6POS code. Of course you need to handle the timing differences / latency but that a different problem..

deadprogram, 2017-05-19:
Hi @gedw99 as you said the same patterns apply, and Gobot also has support for some external hardware devices that themselves provide APIs such as Parrot drones. 

We'd absolutely love to see a contribution to add support for Kuka!

gedw99, 2017-05-19:
https://github.com/nullboundary/goKuka/blob/master/goKuka.go

very simple code gen.
no parametrics as its just a low level code generator.

the isue with kuka is the price. 100 K for a 6d arm that does not need
fencing.

to be frank, i used this stuff 20 years ago in japan on jap bots. Its was
way easier then with a basic serial control interface and not layers of
"enterprise" shiny happy layers in your way. All you wanted was a bot with
a IK engine and basic feedback. Hallo go routines :)

allot of the new comers that are no abb, kuka, etc etc get brought after a
year and their stuff gets swallowed and into the cave of the shiny happy
enterprise layers again.

this one actually lookd good for many reasons. they come from Coro and have
helped on the ROS project immensly with open source abstraction drivers
etc. Now they have a few bots that give the old boys a good kick

http://www.mecademic.com/

http://coro.etsmtl.ca/blog