hybridgroup/gobot #423: Platform Request: Orange Pi Zero

jadr2ddude, 2017-05-26:
I would like to add support for the Orange Pi Zero, and ideally some of the other Orange Pi's. I am planning to use Orange Pi Zero to run a workshop this year, and would ideally like to use gobot for it.

Some helpful links:
[Orange Pi](http://www.orangepi.org/orangepizero/)
[Armbian - the best board image for Orange Pi boards](https://www.armbian.com/)

Armbian has an image with a newer kernel (4.x) for the Orange Pi, and I believe that sysfs access to GPIO works. I will check tommorow on one of my Orange Pi One boards.

deadprogram, 2017-05-26:
Hi @jadr2ddude this post has the GPIO mappings for this board: https://kaspars.net/blog/linux/orange-pi-zero-gpio

Seems like adding it would not take long. Are you planning on working on this? :)

jadr2ddude, 2017-05-26:
@deadprogram yes I will work on it. However I only have Orange Pi One at the moment. I will need to wait for it to be shipped before I can test Zero pin mappings.

deadprogram, 2017-05-26:
This appears to provide the Orange Pi One mappings if you wanted to add that as well: https://cdn.instructables.com/FVP/79LM/INRJEEUJ/FVP79LMINRJEEUJ.LARGE.jpg

chneau, 2018-04-28:
Hello, I'm willing to help for the Orange Pi One [(H3 allwiner)](http://www.orangepi.org/orangepione/) I don't remember my source but here is a json of the mapping.  

```
[
  {
    "key": 3,
    "value": 12
  },
  {
    "key": 5,
    "value": 11
  },
  {
    "key": 7,
    "value": 6
  },
  {
    "key": 8,
    "value": 13
  },
  {
    "key": 10,
    "value": 14
  },
  {
    "key": 11,
    "value": 1
  },
  {
    "key": 12,
    "value": 110
  },
  {
    "key": 13,
    "value": 0
  },
  {
    "key": 15,
    "value": 3
  },
  {
    "key": 16,
    "value": 68
  },
  {
    "key": 18,
    "value": 71
  },
  {
    "key": 19,
    "value": 64
  },
  {
    "key": 21,
    "value": 65
  },
  {
    "key": 22,
    "value": 2
  },
  {
    "key": 23,
    "value": 66
  },
  {
    "key": 24,
    "value": 67
  },
  {
    "key": 26,
    "value": 21
  },
  {
    "key": 27,
    "value": 19
  },
  {
    "key": 28,
    "value": 18
  },
  {
    "key": 29,
    "value": 7
  },
  {
    "key": 31,
    "value": 8
  },
  {
    "key": 32,
    "value": 200
  },
  {
    "key": 33,
    "value": 9
  },
  {
    "key": 35,
    "value": 10
  },
  {
    "key": 36,
    "value": 201
  },
  {
    "key": 37,
    "value": 20
  },
  {
    "key": 38,
    "value": 198
  },
  {
    "key": 40,
    "value": 199
  }
]
```

jadr2ddude, 2018-04-29:
I feel like we could simplify this by just making a multipurpose struct for a SBC where we can just provide a mapping.

gen2thomas, 2025-03-21:
https://www.armbian.com/orange-pi-zero/