hybridgroup/gobot #424: Humidity is disabled in the BME280 I2C driver because of a missing write operation

ssorc, 2017-05-29:
Calling the `Humidity()` function on the `BME280Driver` I2C driver results in the error: "Humidity disabled".

This is because the `initHumidity()` function is incomplete; it writes out the humidity data acquisition options, but doesn't then write to the "ctrl_meas" register, as per Section 5.4.3 of the BME280 data sheet:

> The “ctrl_hum” register sets the humidity data acquisition options of the device. Changes to
> this register only become effective after a write operation to “ctrl_meas”.

I have a patch so can generate a pull request if someone else can confirm that this is the case! I have tested this on a Raspberry Pi Model B using the Sparkfun BME280 Atmospheric Sensor Breakout with gobot 1.5.0 and golang 1.7.5.

deadprogram, 2017-05-30:
Hi @ssorc thanks for the report. I will setup to test this issue shortly and get back to you.

deadprogram, 2017-05-30:
Hi @ssorc yes, I can confirm that error is being returned.

Interestingly, restarting the program causes the readings to come thru correctly on subsequent runs.

In any case, your PR to address this would be appreciated. Thank you!

ssorc, 2017-05-30:
The restart may work because when the call to `d.initialization()` in `Start()` is writing to the `ctrl_meas` register (see the `bmp280RegisterControl` register in the BPM280 driver's `initialization()` function) and this may be enough to finally enable the humidity sensor.

I will send through a PR: it's only a couple of lines.

ssorc, 2017-05-31:
I've submitted the PR.

deadprogram, 2017-06-15:
Version 1.6.0 is now out, and includes this fix. Now closing, thanks again @ssorc for this!

closed 2017-06-15 by deadprogram