hybridgroup/gobot #429: Having issues installing firmata platform features into Win10 system

JDogHerman, 2017-06-14:
I am attemting to install the firmata features following the steps listed [here](https://gobot.io/documentation/platforms/arduino/) and I am getting an error about not enough arguments. 

```
go get -d -u gobot.io/x/gobot/... && go install gobot.io/x/gobot/platforms/firmata
# gobot.io/x/gobot/sysfs
go\src\gobot.io\x\gobot\sysfs\i2c_device.go:88: undefined: syscall.SYS_IOCTL
go\src\gobot.io\x\gobot\sysfs\i2c_device.go:102: undefined: syscall.SYS_IOCTL
go\src\gobot.io\x\gobot\sysfs\i2c_device.go:219: undefined: syscall.SYS_IOCTL
go\src\gobot.io\x\gobot\sysfs\syscall.go:34: not enough arguments in call to syscall.Syscall
        have (uintptr, uintptr, uintptr, uintptr)
        want (uintptr, uintptr, uintptr, uintptr, uintptr)
```
The issue seems to be with the 2nd command ``go install`` as the ``go get`` command run alone does not show an error. My version of windows is Windows 10 Pro v1607 Build 14393.1198

deadprogram, 2017-06-14:
All you need is the:

```
go get -d -u gobot.io/x/gobot/...
```

Docs were already updated here, just awaiting the next release: https://github.com/hybridgroup/gobot/blob/dev/platforms/firmata/README.md#how-to-install

Thanks for the report.

deadprogram, 2017-06-15:
Version 1.6.0 is now out, and includes documentation fixes for this. Now closing, please reopen if needed. Thank you!

closed 2017-06-15 by deadprogram

miko-code, 2017-07-05:
IM having the same issue on win 10  i did the go get as you can see.

but I'm receiving the same error as above 

>  PS C:\projects\go\src\temp> go build
> # gobot.io/x/gobot/platforms/ble
> ..\gobot.io\x\gobot\platforms\ble\ble_client_adaptor.go:189: undefined: defaultDevice
> # gobot.io/x/gobot/sysfs
> ..\gobot.io\x\gobot\sysfs\i2c_device.go:88: undefined: syscall.SYS_IOCTL
> ..\gobot.io\x\gobot\sysfs\i2c_device.go:102: undefined: syscall.SYS_IOCTL
> ..\gobot.io\x\gobot\sysfs\i2c_device.go:219: undefined: syscall.SYS_IOCTL
> ..\gobot.io\x\gobot\sysfs\syscall.go:34: not enough arguments in call to syscall.Syscall
>         have (uintptr, uintptr, uintptr, uintptr)
>         want (uintptr, uintptr, uintptr, uintptr, uintptr)
> 

GO  GET 
> C:\WINDOWS\system32>go get -v -d -u gobot.io/x/gobot/...
Fetching https://gobot.io/x/gobot?go-get=1
Parsing meta tags from https://gobot.io/x/gobot?go-get=1 (status code 200)
get "gobot.io/x/gobot": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot?go-get=1
gobot.io/x/gobot (download)
github.com/hashicorp/go-multierror (download)
Fetching https://gobot.io/x/gobot/api?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/api?go-get=1 (status code 404)
github.com/bmizerany/pat (download)
Fetching https://gobot.io/x/gobot/api/robeaux?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/api/robeaux?go-get=1 (status code 404)
Fetching https://gobot.io/x/gobot/cli?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/cli?go-get=1 (status code 404)
github.com/codegangsta/cli (download)
Fetching https://gobot.io/x/gobot/drivers/aio?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/drivers/aio?go-get=1 (status code 200)
get "gobot.io/x/gobot/drivers/aio": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/drivers/aio?go-get=1
get "gobot.io/x/gobot/drivers/aio": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot?go-get=1
Parsing meta tags from https://gobot.io/x/gobot?go-get=1 (status code 200)
Fetching https://gobot.io/x/gobot/drivers/gpio?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/drivers/gpio?go-get=1 (status code 200)
get "gobot.io/x/gobot/drivers/gpio": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/drivers/gpio?go-get=1
get "gobot.io/x/gobot/drivers/gpio": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/drivers/i2c?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/drivers/i2c?go-get=1 (status code 200)
get "gobot.io/x/gobot/drivers/i2c": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/drivers/i2c?go-get=1
get "gobot.io/x/gobot/drivers/i2c": verifying non-authoritative meta tag
github.com/sigurn/crc8 (download)
github.com/sigurn/utils (download)
Fetching https://gobot.io/x/gobot/sysfs?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/sysfs?go-get=1 (status code 200)
get "gobot.io/x/gobot/sysfs": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/sysfs?go-get=1
get "gobot.io/x/gobot/sysfs": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/examples?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/examples?go-get=1 (status code 404)
Fetching https://gobot.io/x/gobot/platforms/raspi?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/platforms/raspi?go-get=1 (status code 200)
get "gobot.io/x/gobot/platforms/raspi": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/platforms/raspi?go-get=1
get "gobot.io/x/gobot/platforms/raspi": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/gobottest?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/gobottest?go-get=1 (status code 404)
Fetching https://gobot.io/x/gobot/platforms/audio?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/platforms/audio?go-get=1 (status code 200)
get "gobot.io/x/gobot/platforms/audio": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/platforms/audio?go-get=1
get "gobot.io/x/gobot/platforms/audio": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/platforms/beaglebone?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/platforms/beaglebone?go-get=1 (status code 200)
get "gobot.io/x/gobot/platforms/beaglebone": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/platforms/beaglebone?go-get=1
get "gobot.io/x/gobot/platforms/beaglebone": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/platforms/ble?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/platforms/ble?go-get=1 (status code 200)
get "gobot.io/x/gobot/platforms/ble": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/platforms/ble?go-get=1
[... 132 lines cut ...]
[... cut ...]
get "gobot.io/x/gobot/platforms/sphero/ollie": verifying non-authoritative meta tag
Fetching https://gobot.io/x/gobot/platforms/tinkerboard?go-get=1
Parsing meta tags from https://gobot.io/x/gobot/platforms/tinkerboard?go-get=1 (status code 200)
get "gobot.io/x/gobot/platforms/tinkerboard": found meta tag main.metaImport{Prefix:"gobot.io/x/gobot", VCS:"git", RepoRoot:"https://github.com/hybridgroup/gobot"} at https://gobot.io/x/gobot/platforms/tinkerboard?go-get=1
get "gobot.io/x/gobot/platforms/tinkerboard": verifying non-authoritative meta tag