hybridgroup/gobot #430: Neopixel RGB LED strip support?

hairyhenderson, 2017-06-17:
I'm wondering if it'd be possible to add support for NeoPixel-type RGB LED strips (https://www.adafruit.com/product/1376) - these are basically just arrays of WS2812 or SK6812 individually-addressable RGB LEDs.

There are some Raspberry-Pi specific libraries, and there's also an Arduino library, but I'm looking for something more generic that I could use with an ESP8266 over WiFi with Firmata.

Or if there's a different approach I could use, I'm open to options 🙂

hairyhenderson, 2017-06-25:
> What do you think?

I think this is awesome! Thanks @deadprogram 😁

I'll try to test this out too once I get a little free time.

joek, 2017-10-09:
Hi,
what is the status about this project? I am looking for something similar. Do you have the Arduino code, you where using somewhere? It seems the StdFirmata will not work.

One thought about the GPIO Version. There are some problems and limitations to raspberry, which makes it more complicated to implement. Details: https://learn.adafruit.com/neopixels-on-raspberry-pi/overview

deadprogram, 2017-10-10:
Hi @joek you are correct, I have a repos with some work in progress on the Arduino side of things.

https://github.com/hybridgroup/FirmataNeopixels 

Sorry about the complete lack of docs, as of yet. It is a pluging for configurablefirmata if that helps get you started before I can update the READMEs.

You are indeed correct about the speed issues with the raspi, but I expect that for at least a very short length of LEDs a Golang sysfs bitbanging approach might be viable.

rucuriousyet, 2018-05-18:
any updates? I would love to use the raspberry pi with Gobot and some neopixels...

deadprogram, 2018-05-18:
Hi @hairyhenderson your best bet with Raspberry Pi is probably to use a SPI based LED strip like the APA102 https://github.com/hybridgroup/gobot/blob/master/drivers/spi/apa102.go the speed of the Raspi bus is not really enough to drive NeoPixels without an attached microcontroller.

n00dles, 2018-05-31:
Hey Guys, any update on getting this working? 

Thought I had it going (1 pixel lit up for me) but I haven't been able to replicate. 

When adding to the esp firmware do I need to add : 

 firmataExt.addFeature(Neopixels);

or similar? to the initFirmata() function.

n00dles, 2018-05-31:
ok figured I did need to use addFeature and I get lights !!

but just a flash the I get an exception on the ESP and it goes in to a reboot loop. 
I'm just using your example code. 

any suggestions?

deadprogram, 2018-06-01:
The ESP8266 basically goes into the reboot loop under many circumstances and one in particular is when it does not have enough time to process all the WiFi data. I ran into that exact problem with it, and was not even using code that is so timing dependent as the Neopixels. My Neopixel experiments were using an Intel TinyTile and using BLE not WiFi, as I recall.

One thought was to switch over to using Homie on the ESP8266 https://github.com/marvinroger/homie-esp8266 and use an MQTT based protocol. I have done some experiments with defining a Firmata-like protocol on top of Homie but it is far from complete.

Another useful approach might be to do the same sort of thing as Homie, but using COAP instead of MQTT. That would have quite a few benefits, such as not requiring a server, and also allowing for pure browser-based clients to test the device, but I have not seen that anyone has done anything even close to that yet. It is just one of several ideas I have been playing with.

n00dles, 2018-06-01:
@deadprogram Thanks for the update and advise. 

I have it working now on wifi on esp8266 but I still get the odd reboot after about a minute. so I'll keep at it.

I'm so close to having it for what I want, not fancy animations just to be able to light up sequences of leds.

deadprogram, 2018-06-01:
That is good news. Please update this issue with details once you get it working to your liking.

n00dles, 2018-06-03:
quick question. 

what is the purpose of the lines 

[code]
   Firmata.write(START_SYSEX);
   Firmata.write(NEOPIXEL_CMD);
   Firmata.write(PIXEL_SHOW);
   Firmata.write(END_SYSEX);
[/code]

in the arduino code? 

Removing it seems to get rid of any exceptions I'm getting. is this required for proper operation?

deadprogram, 2018-06-04:
That is the expected Firmata response sent from the Arduino back to the caller. The exception is on the ESP8266 side or the Go side?

n00dles, 2018-06-04:
Exception is on the Arduino (esp8266) side. 

without the above lines I can send pixel data for hours. 
with the lines included it crashes within a minute. 

Crash dump below. 

`Exception: 9 (LoadStoreAlignmentCause: Load or store to an unaligned address)

epc1:     0x4022e4b5: tcp_write at core/tcp_out.c:586
epc2:     0x00000000
epc3:     0x00000000
excvaddr: 0x00110009
depc:     0x00000000

ctx: cont

sp:       0x3ffefb50
end:      0x3ffefeb0
offset:   0x000001a0

stack:
0x4022e4d8: tcp_write at core/tcp_out.c:626
0x4022e484: tcp_write at core/tcp_out.c:575
0x40203472: ClientContext::_write_some() at ??:?
0x40203514: ClientContext::_write_from_source(DataSource*) at ??:?
0x40202651: loop at ??:?
0x402035c5: WiFiClient::write(unsigned char const*, unsigned int) at ??:?
0x402060c4: WiFiClient::connect(String, unsigned short) at ??:?
0x40205921: _GLOBAL__sub_I__ZN9IPAddressC2Ev at IPAddress.cpp:?
0x4020277c: WiFiStream::write(unsigned char) at ??:?
0x4020638d: FirmataClass::sendDigitalPort(unsigned char, int) at ??:?
0x40204e1f: FirmataNeopixels::setConfig(unsigned char, unsigned short) at ??:?
0x40204f4d: FirmataNeopixels::handleSysex(unsigned char, unsigned char, unsigned char*) at ??:?
0x40204c3d: FirmataExt::handleSysex(unsigned char, unsigned char, unsigned char*) at ??:?
0x4020613c: WiFiClient::connected() at ??:?
0x40204cc3: handleSysexCallback(unsigned char, unsigned char, unsigned char*) at ??:?
0x40204140: FirmataClass::processSysexMessage() at ??:?
0x40204260: FirmataClass::parse(unsigned char) at ??:?
0x402043ca: FirmataClass::processInput() at ??:?
0x40202660: loop at ??:?
0x40205e40: esp_schedule at ??:?
0x40100700: cont_norm at cont.S.o:?`

n00dles, 2018-06-04:
Exception is on the arduino (esp8266) 

Without the above line I can send data to the device for hours without issue. 
With the lines the device crashes within seconds. 

Dump below. 

`Exception: 9 (LoadStoreAlignmentCause: Load or store to an unaligned address)

epc1:     0x4022e374: tcp_write at core/tcp_out.c:427 (discriminator 1)
epc2:     0x00000000
epc3:     0x00000000
excvaddr: 0x002b0001
depc:     0x00000000

ctx: cont

sp:       0x3ffefb30
end:      0x3ffefeb0
offset:   0x000001a0

stack:
0x4022e538: tcp_write at core/tcp_out.c:633
0x4022e4e4: tcp_write at core/tcp_out.c:586
0x40203472: ClientContext::_write_some() at ??:?
0x40203514: ClientContext::_write_from_source(DataSource*) at ??:?
0x402035c5: WiFiClient::write(unsigned char const*, unsigned int) at ??:?
0x40206124: WiFiClient::write(unsigned char) at ??:?
0x40205981: Print::write(char const*) at ??:?
0x4020277c: WiFiStream::write(unsigned char) at ??:?
0x402063ed: FirmataClass::write(unsigned char) at ??:?
0x40204e1f: FirmataNeopixels::setPixel(unsigned short, unsigned int) at ??:?
0x40204f9b: FirmataNeopixels::handleSysex(unsigned char, unsigned char, unsigned char*) at ??:?
0x40100690: free at ??:?
0x40204c3d: FirmataExt::handleSysex(unsigned char, unsigned char, unsigned char*) at ??:?
0x4020619c: WiFiClient::operator bool() at ??:?
0x40204cc3: handleSysexCallback(unsigned char, unsigned char, unsigned char*) at ??:?
0x40204140: FirmataClass::processSysexMessage() at ??:?
0x40204260: FirmataClass::parse(unsigned char) at ??:?
0x402043ca: FirmataClass::processInput() at ??:?
0x40202660: loop at ??:?
0x40205ea0: loop_wrapper() at core_esp8266_main.cpp:?
0x40100700: cont_norm at cont.S.o:?`

n00dles, 2018-06-04:
ignore the above, I was doing something stupid....

NomadMaker, 2018-06-07:
It would probably be a lot easier to use DotStar strips than NeoPixel ones. DotStar is AdaFruit's name for LED's controlled via the APA102. They don't have the strict speed requirement that the NeoPixel ones do.

I have both DotStar and NeoPixel things, but I can't get to them for the next week or so because I'm packing up my motorhome and moving it from Florida to Pennsylvania shortly. If I can find a DotStar strip fast enough then I will see if I can test it when I'm stopped for the night. However, they may be already boxed and put away and I don't have the time to search for them until I get to my destination.

I am planning on eventually replacing the LED strips on my RV with DotStar strips so that I can be ready for holidays with a simple command. Imagine various types of Christmas lighting, then 4th of July, then just basic white, then Halloween, then Easter. That way nobody can make me change lights. :)