hybridgroup/gobot #432: ADAFRUIT 16-CHANNEL 12-BIT PWM/SERVO DRIVER - I2C INTERFACE driver implementation

chaitanya9186, 2017-06-27:


chaitanya9186, 2017-06-27:
Overview and details: https://learn.adafruit.com/16-channel-pwm-servo-driver?view=all

deadprogram, 2017-06-27:
Hi @chaitanya9186 check out the `dev` branch where we have just recently done work on this driver:
https://github.com/hybridgroup/gobot/blob/dev/drivers/i2c/pca9685_driver.go

Example code here:
https://github.com/hybridgroup/gobot/blob/dev/examples/firmata_pca9685.go

chaitanya9186, 2017-06-27:
Awesome @deadprogram.

chaitanya9186, 2017-06-27:
Can I use this driver from Raspberry Pi as well?

deadprogram, 2017-06-27:
Yes, it is just an I2C device.

chaitanya9186, 2017-06-27:
I am pretty new to this. 

In the [example](https://github.com/hybridgroup/gobot/blob/dev/examples/firmata_pca9685.go), it is using [firmata platform](https://github.com/hybridgroup/gobot/tree/dev/platforms/firmata). If I want to use in Raspberry Pi, I need to use [raspi platform](https://github.com/hybridgroup/gobot/tree/dev/platforms/raspi)

deadprogram, 2017-06-28:
Yes, that is correct. You would just use the raspi package and Adaptor instead of the firmata one. SImilar to this example: https://github.com/hybridgroup/gobot/blob/master/examples/raspi_blinkm.go#L17-L18

Hope that helps!

deadprogram, 2017-07-06:
I'm going to leave this issue open until the PCA9685 driver makes it to the master branch. Once it has been merged, I'll come back and close this issue unless further assistance is needed.

deadprogram, 2017-07-31:
Good news is this was merged into master a couple weeks ago with the release of v1.6.1 however, the needed documentation is lagging a little behind. I'll continue to leave this open until the docs update.

deadprogram, 2017-11-15:
A example of use and docs page has been added to site. Now closing this issue, thanks everyone.

closed 2017-11-15 by deadprogram