hybridgroup/gobot #434: doesn't compile on windows

amits1995, 2017-07-04:
I get the following compilation error while trying to run the simple blink example for arduino:  
```
# gobot.io/x/gobot/platforms/ble
..\go\src\gobot.io\x\gobot\platforms\ble\ble_client_adaptor.go:189: undefined: defaultDevice
# gobot.io/x/gobot/sysfs
..\go\src\gobot.io\x\gobot\sysfs\i2c_device.go:88: undefined: syscall.SYS_IOCTL
..\go\src\gobot.io\x\gobot\sysfs\i2c_device.go:102: undefined: syscall.SYS_IOCTL
..\go\src\gobot.io\x\gobot\sysfs\i2c_device.go:219: undefined: syscall.SYS_IOCTL
..\go\src\gobot.io\x\gobot\sysfs\syscall.go:34: not enough arguments in call to syscall.Syscall
	have (uintptr, uintptr, uintptr, uintptr)
	want (uintptr, uintptr, uintptr, uintptr, uintptr)
exit status 2
Process exiting with code: 1
```

Here's the actual code:
```
package main

import (
        "time"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/firmata"
)

func main() {
        firmataAdaptor := firmata.NewAdaptor("COM8")
        led := gpio.NewLedDriver(firmataAdaptor, "13")

        work := func() {
                gobot.Every(1*time.Second, func() {
                        led.Toggle()
                })
        }

        robot := gobot.NewRobot("bot",
                []gobot.Connection{firmataAdaptor},
                []gobot.Device{led},
                work,
        )

        robot.Start()
}
```

deadprogram, 2017-07-06:
Hello @amits1995 this was an unexpected consequence of adding support for BLEFirmata, since there is not yet Windows support in the BLE library. 

Thanks for reporting. I'll look into either making that dependency conditional, or some other soution.

eriawan06, 2017-07-07:
hallo @deadprogram, I also encountered problem like this. How to fix it?

tural-esger, 2017-07-10:
The same issue with me too.

deadprogram, 2017-07-12:
Hello, everyone.

Just committed code to the `dev` branch that corrects this issue. 

In addition, I have now created a Windows-based build in Appveyor to hopefully help avoid sudden surprises in the future. 

Sorry about the inconvenience, and thanks everyone who reported this issue with good debugging details.

This issue will remain open for comment until the code gets to `master` in this week's release.

deadprogram, 2017-07-18:
Hi, everyone

I released v1.6.1 a few days ago. This issue has now been resolved, so I am now closing. Thanks again to everyone who helped out!

closed 2017-07-18 by deadprogram