hybridgroup/gobot #436: An example to convert i2c drivers written in C/C++ and Python to golang

chaitanya9186, 2017-07-05:
Hi

I would like to work on #435 (Implement LSM9DS0 device driver). 

I have [reference code](https://github.com/adafruit/Adafruit_LSM9DS0_Library) available in C++. Is there anyway I can convert this straight into gobot i2c device driver. 

Please provide me if there are any examples.

Thank you.

deadprogram, 2017-07-06:
Although it is possible to wrap C/C++ code in Golang, all of our i2c drivers, and most of the code that we can, is written in pure Golang to make it easier to cross-compile, and reduce dependencies.

We would normally just examine the C++ reference code, and "manually transpile" it into Golang. It would be ideal if we had some docs on this, or someone wrote a blog post.

gen2thomas, 2023-05-15:
I'm going to close this issue, because there are programs available, which can transform code to golang. Also enough documentation is out there, so we do not reinvent the wheel here.

closed 2023-05-15 by gen2thomas