hybridgroup/gobot #439: raspberry pi built-in pull up/down resistor support?

boundedinfinity, 2017-07-18:
Is there a way to configure the raspberry pi internal pull up/down resistors?  I'm looking for help (or just a pointer to documentation) do some something similar to the configuration of the GPIO pin's pull up/down resistor in the python [RPi.GPIO](https://sourceforge.net/p/raspberry-gpio-python/wiki/BasicUsage/) module.

In python the configuration is achieved by something like this:

```python
GPIO.setup(PIN, GPIO.IN, pull_up_down = GPIO.PUD_DOWN)
```

Do gobot have something similar?

spatil, 2017-11-02:
Is there any workaround for this?

gokuu, 2019-12-21:
I have the same issue :(

stuartleeks, 2020-04-25:
@deadprogram - do you have any sense of what would be needed to add this? (From what I can tell it isn't currently supported?)

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas