hybridgroup/gobot #440: Why are we not using golang.org/x/exp/io/i2c Interface for define i2c devices?

chaitanya9186, 2017-07-20:
I have seen [golang.org/x/exp/io/i2c](https://github.com/golang/exp/blob/master/io/i2c/i2c.go), which looks most generic for the purpose of i2c interface.

1. Why did we define new [i2c interface](https://github.com/hybridgroup/gobot/blob/dev/drivers/i2c/i2c.go) when we have a one available in golang exp lib?
    What I mean is, how is the current one it different from experimental lib?
2. What are the advantages?
3. Is there any way to convert current i2c interface to use the experimental?

Thank you

deadprogram, 2017-07-21:
Hi @chaitanya9186 so the Gobot interfaces predate that work by a couple of years. Also, x/exp/io worked is not actually being developed on any longer, as you might notice from the commits. :(

The interfaces are largely equivalent. Gobot is intended to work with both embedded Linux as well as microcontrollers connected via various serial or wireless adaptors, so the interface design is informed to some extent by things like Arduino etc.

If the community at large wants to start supporting the x/exp/io interfaces, the internal implementations of Gobot could evolve in that direction. However, so far, that has not been the case.

Hope that helps clarify a bit.

chaitanya9186, 2017-07-21:
Thanks @deadprogram.

closed 2017-07-21 by chaitanya9186