hybridgroup/gobot #441: Trying to run joystick example

kc1116, 2017-07-21:
Getting this error when trying to run the joystick example is the example outdated now ?
./robot.go:17: undefined: gobot.On

deadprogram, 2017-07-21:
Hello @kc1116 can you please point us to the specific example so we can update it? You are most certainly correct about it being out of date.

kc1116, 2017-07-21:
This example is outdated https://github.com/hybridgroup/gobot/tree/master/platforms/joystick, it is also on the gobot.io website. Also it might be worth mentioning on macOS in order to get gobot to detect my xbox joystick I had to install https://github.com/360Controller/360Controller your joystick won't get detected automatically without some sort of driver for it.

deadprogram, 2017-07-25:
HI @kc1116 please check out this commit https://github.com/hybridgroup/gobot/commit/1312c20d4be94fde9b600ce5976effde3c1d9b50 to the `dev` branch. I think it should address most of your issue.

I am going to reopen this, please close if you think it has been handled, or provide additional feedback.

Thanks!

kc1116, 2017-07-26:
Does gobot support anything for transceiving lasers or something to that extent ?

deadprogram, 2017-07-27:
You mean like using a laser to send or receive a signal? Not that I have seen.