hybridgroup/gobot #444: Ardrone camera

kc1116, 2017-07-30:
Is there any plan to add support for the parrot Ardrone 2.0 camera, if not can you point me in the right direction to add this functionality myself?

deadprogram, 2017-07-31:
Hi @kc1116 we do have some support for the ARDrone 2 video stream, by using OpenCV. 

Here is an example: https://github.com/hybridgroup/gobot/blob/master/examples/ardrone_face_tracking.go

That might be what you are looking for?

kc1116, 2017-07-31:
I've tried to run this example but the go-opencv library keeps giving a index out of range error. The bebop supports streaming to ffmpeg, is there a PaVE parser somewhere in this code I can use to manipulate the video stream manually ?

deadprogram, 2017-08-02:
No parser yet, but probably several places to find code to write one.

Can you please provide a stack trace or other info for your opencv problem?

kc1116, 2017-08-02:
I am currently working on a parser will create a branch and have you look at it.

gen2thomas, 2024-02-21:
Hi @kc1116 is there any progress on this topic or a branch exist? Is that still of interest?