hybridgroup/gobot #445: Support for "retain" in MQTT messages

Remo, 2017-08-11:
I'm trying to send a message to an MQTT broker, but retain it. I did some digging and it seems like paho already supports this. The third parameters sets it to `false`: https://github.com/hybridgroup/gobot/blob/master/platforms/mqtt/mqtt_adaptor.go#L124

I'm new to go and gobot and thus wasn't able to create a pull request. Not sure how I'm uspposed to add this without optional arguments or method overloading in a way that doesn't break backwards compatibility.

Also, why not return the result of `a.client.Publish`  so that we can track the delivery of the message?

deadprogram, 2017-08-28:
Hi @Remo thanks for the comments. Seems like a good idea to implement.

Regarding optional arguments, the best way to handle this is using variadic configuration ala https://dave.cheney.net/2014/10/17/functional-options-for-friendly-apis

Gobot's i2c drivers all already do this, to provide a slightly more in-context example: https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/bme280_driver.go#L42

Returning a paho.Token from Publish() certainly seems like a small but useful change, as well.

We'd be very happy to accept a PR with these changes. Please let me know if I can help out in any way.

Remo, 2017-08-28:
@deadprogram thanks! I'm still evaluating, not sure I'll spend time learning go and gobot at this point. Currently playing around with python libraries and framework.

andrewebdev, 2019-07-19:
@Remo I've just submitted a PR that should allow you to access `paho.Client` directly from the gobot `mqtt.Adaptor`.

See andrewebdev:682-public-paho-client

gen2thomas, 2022-09-19:
@deadprogram it seems the originated problem is already fixed with the merged PR #719 . Maybe you could close the issue?

closed 2022-09-19 by Remo