hybridgroup/gobot #446: ssd1306_driver

qarth, 2017-08-20:
How do you display text with the ssd1306_driver driver?

deadprogram, 2017-08-28:
Hi @qarth right now the functionality is rather limited. 

Once a SPI interface has landed into Gobot dev, a more performant interface to the SSD1306 will be possible, and drawing text a more reasonable prospect.

deadprogram, 2017-09-11:
I think that @ulisesflynn is going to work on the SPI version of this driver in the near future. :smile_cat:

ulisesflynn, 2017-09-11:
I've been pretty busy this past week, but I'll get to this soon!

mikegleasonjr, 2017-10-22:
Here's an example displaying the current time:

```
package main

import (
	"image"
	"time"

	"github.com/fogleman/gg"
	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/i2c"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	r := raspi.NewAdaptor()
	d := i2c.NewSSD1306Driver(r)

	work := func() {
		d.Clear()
		d.Display()

		ctx := gg.NewContext(d.Buffer.Width, d.Buffer.Height)

		gobot.Every(time.Second, func() {
			ctx.SetRGB(0, 0, 0)
			ctx.Clear()
			ctx.SetRGB(1, 1, 1)
			ctx.DrawStringAnchored(time.Now().Format("15:04:05"), 0, 0, 0, 1)
			transfer(d, ctx.Image())
		})
	}

	robot := gobot.NewRobot("blinkBot",
		[]gobot.Connection{r},
		[]gobot.Device{d},
		work,
	)

	robot.Start()
}

func transfer(d *i2c.SSD1306Driver, img image.Image) {
	d.Clear()
	for y, w, h := 0, img.Bounds().Dx(), img.Bounds().Dy(); y < h; y++ {
		for x := 0; x < w; x++ {
			c := img.At(x, y)
			r, _, _, _ := c.RGBA()
			if r == 0 {
				continue
			}
			d.Set(x, y, 1)
		}
	}
	d.Display()
}
```
Basically I render an image in memory using a 3rd party library, then I set the pixels one by one of that image on the ssd1306.

deadprogram, 2017-10-22:
Hi @mikegleasonjr that transfer looks useful. Perhaps a version of that function `TransferImage` could be added to the driver itself? I'd gladly accept such a PR.

mikegleasonjr, 2017-10-22:
@deadprogram Great, will happily do!

deadprogram, 2018-04-16:
Hi @mikegleasonjr any chance of still getting this PR? :smile_cat:

deadprogram, 2018-05-31:
Now closing this issue since ssd1306 support is in fact in Gobot. Still love to see a PR with the image transfer feature.

closed 2018-05-31 by deadprogram