hybridgroup/gobot #449: Rock64

jerson, 2017-09-08:
rock64 support from pine64.org https://www.pine64.org/?page_id=7147

resources:
http://files.pine64.org/doc/rock64/ROCK64_Pi-2%20_and_Pi_P5+_Bus.pdf
http://files.pine64.org/doc/rock64/Rockchip_RK805_Datasheet_V1.1%C2%A020160921.pdf
http://files.pine64.org/doc/rock64/ROCK64_Schematic_v2.0_20170704.pdf

deadprogram, 2017-09-09:
This is a duplicate of https://github.com/hybridgroup/gobot/issues/362

jerson, 2017-09-09:
rock64 is different than pine64 board

gen2thomas, 2023-05-18:
Meanwhile I own a Pine Rock64 v2.0, so I reopen this issue as a reminder.

gen2thomas, 2023-05-18:
Note: RockPro64 is a different one with RK3399. I'm running this with libreELEC, but [armbian](https://forum.pine64.org/showthread.php?tid=6928&page=6) should also be possible.

gen2thomas, 2023-12-13:
I reopen this issue as a reminder for me.

gen2thomas, 2025-02-16:
fixed in v2.5.0

closed 2025-02-16 by gen2thomas