hybridgroup/gobot #451: MQTT is cleaning up session after auto reconnect

joek, 2017-09-24:
Hi, 
I got a problem, if the mqtt adaptor is reconnecting. The subscriptions are not recreated by the adaptor. 

There are two posibilites to deal with it. 
1. We could use the [OnReconnectHandler](https://godoc.org/github.com/eclipse/paho.mqtt.golang#ClientOptions.SetOnConnectHandler)
2. By default the session is cleaned up on every connect. By setting [cleanSession](https://godoc.org/github.com/eclipse/paho.mqtt.golang#NewClientOptions) to false, resubscribing is not needed any more. 

I implemented a setter and getter for option 2. If you appreciate a pull request I would love to create one.

https://github.com/joek/gobot/commit/5be2c8a60a0049d0c67d68e139f3281e308b1d62

deadprogram, 2017-09-24:
Hello @joek that seems like a great thing, please submit a PR, and thank you!

deadprogram, 2017-10-23:
This has been released as part of https://github.com/hybridgroup/gobot/releases/tag/v1.7.0 so I am now closing. Thanks again, @joek

closed 2017-10-23 by deadprogram