hybridgroup/gobot #453: gpiochip458 base

BakeRolls, 2017-10-05:
Hi, I am trying to use Gobot with gokrazy (on a Raspberry Pi 3). To get GPIO working, adding `CONFIG_GPIO_SYSFS=y` to the [`.config`](https://github.com/gokrazy/kernel/blob/master/cmd/gokr-build-kernel/build.go#L19) is enough. The only problem is, that the pin indices don't start with 0 but 458:

```bash
/ # ls /sys/class/gpio/
export       gpiochip458  unexport
/ # cat /sys/class/gpio/gpiochip458/base 
458
/ # cat /sys/class/gpio/gpiochip458/label 
pinctrl-bcm2835
```

I used [breakglass](https://github.com/gokrazy/breakglass) (as suggested in an [issue on gokrazy](https://github.com/gokrazy/gokrazy/issues/33)) to debug.

Is there a way to, for example, set the base within Gobot (i.e. `pin + $(cat /sys/class/gpio/gpiochip*/base)`) or should pin numbers always start at 0?

gen2thomas, 2022-09-20:
Hi @bake, sorry for the long delay, but it seems there is no knowledge about working with gokrazy. 

I'm not sure about this is:
* a general bug in raspi pin mapping
* a problem on interacting with gokrazy
* a request for a new/special platform (adaptation of Raspi3)

Can someone make it more clear?

bake, 2022-09-29:
Thank you for the reply! I am currently not using gocrazy but it looks like GPIO should be working now.

gen2thomas, 2022-09-29:
@bake thank your for the answer, so I going to close this issue.

closed 2022-09-29 by gen2thomas