hybridgroup/gobot #454: Use dep for dependency management

trevrosen, 2017-10-07:
## What
I'm proposing to use [dep](https://github.com/golang/dep) as a tool to ensure a canonical dependency installation process for development.

## Why
It will make development far smoother and make reproducible builds much easier.

deadprogram, 2017-10-08:
Hi @trevrosen this seems like a great idea to me.

deadprogram, 2017-10-08:
Hello @trevrosen regarding your question about what has been driving the backward version compatibility. We have been trying to ensure support for the current version of Go - 2, but perhaps we should instead mimic the Go languages's own policy: https://golang.org/doc/devel/release.html#policy

deadprogram, 2017-10-08:
This thread describes some specific issues regarding Go language compat. and Gobot: https://github.com/hybridgroup/gobot/issues/390#issuecomment-289812820

deadprogram, 2017-11-15:
Thanks to @trevrosen this is now in the `dev` branch. I will close this issue once it makes it into the next release.

PR #455 (2017-11-15): Add dep, change how tests run in CI
Fixes #454

## What
This PR moves dependency management out of a manually maintained list in the `Makefile` and into dep artifacts (see issue #454). It also moves CI checks from in-lined bash in `.travis.yml` into `Makefile` targets and ensures that tests will exclude code in the `vendor/` directory. It also implements a variety of other build fixes and opinions as detailed below.

## Why
This will make development easier, less error-prone, and enable some dependency re-factoring that I've wanted to do for awhile. More broadly, the changes in this PR ensures [... cut]
commit d959e534 (2017-11-15): dep for building vendor; no vendor in CI tests

 Fixes #454
closed 2017-12-21 by deadprogram