hybridgroup/gobot #456: Use a custom Docker container for builds

trevrosen, 2017-10-07:
## What
Run the Gobot build process inside a custom container

## Why
This will *dramatically* speed up the process, as it will make use of Docker's container image layer cache to ensure that the system lib installation process effectively only has to happen once per active TravisCI Docker host instead of on every single build.

## How
The process is outlined here:https://docs.travis-ci.com/user/docker/

## Who?
I volunteer to do this if @deadprogram is down with it.

deadprogram, 2017-10-08:
This seems like an extremely good idea. Probably should have done this a while ago, or at the very least I should have done that instead of switching to the Travis build cache for the recent changes to support GoCV and OpenCV 3.3.

trevrosen, 2017-10-08:
Ok cool. I've put together a container locally and am figuring out how Travis works when you're using both a container and testing different versions of Go. Ideally it'd let me just install each supported version of Go myself in different tagged containers, b/c then I could move just about everything except for the test run and style vetting into the image, but I need to play around with it a bit.

Do you have a DockerHub account for Gobot?

deadprogram, 2017-12-21:
This was released as part of v1.8.0 thanks again for your awesome work, @trevrosen

trevrosen, 2017-12-21:
Yay! Happy to contribute. All for the love of robots. :-)

trevrosen, 2017-12-21:
This particular piece of the puzzle isn't 100% worked out yet so I'm going to re-open until I can put together the full container and have it leveraged in Travis.

gen2thomas, 2025-02-15:
I'm going to close this issue due to long time of inactivity. Feel free to reopen, if needed.

closed 2025-02-15 by gen2thomas