hybridgroup/gobot #458: i2c raspberry and arduino

muhtadi, 2017-10-18:
how to communicate raspberry and arduino via i2c?

deadprogram, 2017-10-22:
Hi @muhtadi are you wanting to use an Arduino with some specialized functionality as an I2C slave device, and the Raspi as the controller?

muhtadi, 2017-10-22:
yes, please help

deadprogram, 2017-10-23:
You might want to check out the blog post at https://oscarliang.com/raspberry-pi-arduino-connected-i2c/ as far as how to wire up the Arduino to the Raspi. The Gobot side requires either writing a new Driver, or just using the Raspi Adaptor's I2C interface directly.

If you can provide the Arduino sketch I can provide further details on the Gobot side.

muhtadi, 2017-10-25:
// Wire Slave Sender
#include <Wire.h>

void setup() {
  Wire.begin(8);                // join i2c bus with address #8
  Wire.onRequest(requestEvent); // register event
}

void loop() {
  delay(100);
}

// function that executes whenever data is requested by master
// this function is registered as an event, see setup()
void requestEvent() {
  Wire.write("hello "); // respond with message of 6 bytes
  // as expected by master
}

muhtadi, 2017-10-25:
master receiver raspi i2c in gobot code?

deadprogram, 2017-10-26:
Yes, Gobot should act as the master, and performing a Read() on the same address as your arduino, should retrieve data. Check out https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/hmc6352_driver.go for a relatively simple example of an i2c driver.