hybridgroup/gobot #460: Be able to change the resolution of the ssd1306_driver

mikegleasonjr, 2017-10-22:
I am using the mini 128x32 pioled from adafruit (https://www.adafruit.com/product/3527). Right now the default buffer size is hardcoded to 128x64.

It is working fine using the current library, but it causes extra processing having twice the size for no reason.

I can also construct the driver myself just like in `NewSSD1306Driver()`, but an optional param like `WithDisplayBuffer(...)` or `WithWidth()/WithHeight()` would be great.

Thanks fo this amazing library!

deadprogram, 2017-10-22:
HI @mikegleasonjr sounds like you have the right idea on how to handle via `WithDisplayWidth()` or `WithDisplayHeight()` param functions that get passed along to init the `DisplayBuffer`.

I have added an implementation into the `dev` branch that does this, but I currently do not have an OLED driver with any other characteristics then the default. Can you please give this a try, and let me know if it works as you expect?

deadprogram, 2017-10-23:
This has been released as part of https://github.com/hybridgroup/gobot/releases/tag/v1.7.0 so I am going to close this issue. Please reopen if needed.

closed 2017-10-23 by deadprogram