hybridgroup/gobot #461: gpio.ButtonPush & gpio.ButtonRelease are inverted

mikegleasonjr, 2017-10-22:
Hi, I'm using an [adafruit oled](https://learn.adafruit.com/adafruit-128x64-oled-bonnet-for-raspberry-pi?view=all) display with buttons. Following this simple example: https://gobot.io/documentation/examples/raspi_button/, my events are reversed.

Which means I get the `release` event when I push the button, and the `push` event when I release it.

I don't know how the buttons on the hat are wired, can I do something about it beside handling the events that way?

deadprogram, 2017-10-22:
Hi @mikegleasonjr I do not have one of these hats, but was looking thru the docs on the Adafruit site. Seems like the Python code samples expect that the buttons are pulled up.

I have just added some code to the `dev` branch to the `gpio.ButtonDriver` for `DefaultState`. Please try setting `DefaultState = 1` like this:

```go
	d := NewButtonDriver(a, "1")
	d.DefaultState = 1
```

It should reverse the behavior, making a change from LOW TO HIGH trigger a release, and HIGH to LOW trigger a push.

mikegleasonjr, 2017-10-22:
@deadprogram Yup this is working fine now in dev 👍

deadprogram, 2017-10-22:
OK, cool. I'll leave this issue open until this code is released. Thanks for testing.

deadprogram, 2017-10-23:
This is now released as part of https://github.com/hybridgroup/gobot/releases/tag/v1.7.0 so closing. Thanks again!

closed 2017-10-23 by deadprogram