hybridgroup/gobot #467: beaglebone pwm does not works as since uboot overlays there is no slots file

johanhenselmans, 2017-11-16:
Since march 2017 the beaglebone developers have decided to migrate away from kernel overlays to uboot overlays. That means there is no slots file anymore, see 
https://groups.google.com/d/msg/beagleboard/iRUT7UXYjjA/LpG3OLmcBAAJ and https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays. 

I do not understand whtat the ensureSlots func in beaglebone_adaptor.go  is doing, but with the current uboot overlays it does not work. 

Also mentioned is that pwmchip can change, depending on which overlay is used: so one has to look at eg (/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/) to find out which pwmchip name a specific pwm pin has. 

See  https://github.com/RobertCNelson/bb.org-overlays/blob/master/examples/cape-unversal-pwm.txt#L3-L9

If I can be of any help, please let me know.

deadprogram, 2017-11-17:
Hi @johanhenselmans thanks for bringing this up.

The `ensureSlots()` function is intended to initialize the ADC interface. As you point out, that has now been deprecated.

Seems like the best solution here, is probably to use the sysfs and pixmux interfaces directly to configure pins as desired. The page https://github.com/beagleboard/bb.org-overlays/tree/master/tools/beaglebone-universal-io#details breaks down the newer simplified patterns. Seems like we just need to basically do the same thing as `config-pin` does.

So, you seem to be offering to help out, in which case the answer is "yes"!

deadprogram, 2017-11-17:
Looks like Adafruit's Python lib is doing a lot of the right things, as a source of inspiration: https://github.com/adafruit/adafruit-beaglebone-io-python

johanhenselmans, 2017-11-17:
Hmm,  I am not fluent in Gobot/Golang, so spare me some slack...
I'll see what I come up with.

deadprogram, 2017-11-17:
I can help, as well. Always happy to collaborate on improvements.

deadprogram, 2017-12-16:
Hello @johanhenselmans or anyone else interested in the BBB. I have a branch with some work in progress on this located here: https://github.com/hybridgroup/gobot/tree/feature/pocket-beaglebone

Please take a look if you can, and thanks.

johanhenselmans, 2017-12-17:
Interesting, I was working to put the adafruit python code in a go-able version. It's not working nor tested yet. I'll have time to work on it tomorrow.

deadprogram, 2017-12-20:
Code that runs on the latest Beaglebone Debian OS is now in the `dev` branch. I will leave this issue open until the next Gobot release.

deadprogram, 2017-12-26:
This was released with v1.8.0 a few days ago, so now closing. Thanks!

closed 2017-12-26 by deadprogram

johanhenselmans, 2018-02-26:
Ron, I was hoping to see you at Fosdem, but was too influensa'd to get in Brussels. The current version does not take into account that about all the pins can behave as GPIO's, or as PWMs for that matter (with some PRU massaging). I tried to come up with a better solution, but I am having trouble understanding how I should compile my branch so that I can test it properly.  I seem to be missing some step. 

'go build' in the $GOPATH/src/gobot.io/x/gobot does not seem to work, should I build with make? I ran make, but make ran tests, and then gave an error:

--- PASS: TestPwmPinDutyCycleError (0.00s)
PASS
ok  	gobot.io/x/gobot/sysfs	0.405s
make: *** [test] Error 2

How do I build local branches so that I can test locally and contribute afterwards?

johanhenselmans, 2018-06-12:
Ron, I have some problems with it as a bunch of pins that can also act as PWM are not defined. I have tried to come up with a solution, but I am unable to get the stuff compiled properly.

Is there some kind of manual that explains step by step how to build a test version that than can be used for submitting pull requests?

> 
> Closed #467 <https://github.com/hybridgroup/gobot/issues/467>.

Vriendelijke Groeten,

Johan Henselmans
johan@netsense.nl
+31 6 51362675